meshtastic/python #1: Missing files in sdist

thatch, 2020-05-21:
It appears that the manifest is missing at least one file necessary to build
from the sdist for version 0.5.1. You're in good company, about 5% of other
projects updated in the last year are also missing files.

```
+ /tmp/venv/bin/pip3 wheel --no-binary meshtastic -w /tmp/ext meshtastic==0.5.1
Looking in indexes: http://10.10.0.139:9191/root/pypi/+simple/
Collecting meshtastic==0.5.1
  Downloading http://10.10.0.139:9191/root/pypi/%2Bf/8ed/52fe4d06cb0e4/meshtastic-0.5.1.tar.gz (11 kB)
    ERROR: Command errored out with exit status 1:
     command: /tmp/venv/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-wheel-jd7i0a45/meshtastic/setup.py'"'"'; __file__='"'"'/tmp/pip-wheel-jd7i0a45/meshtastic/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-wheel-jd7i0a45/meshtastic/pip-egg-info
         cwd: /tmp/pip-wheel-jd7i0a45/meshtastic/
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-wheel-jd7i0a45/meshtastic/setup.py", line 7, in <module>
        with open("README.md", "r") as fh:
    FileNotFoundError: [Errno 2] No such file or directory: 'README.md'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

thatch, 2020-05-21:
This appears to work with pip 20, but not pip 19.

geeksville, 2020-06-17:
@thatch Thanks for the report.  Sorry for the late reply, I had somehow not had "Watch" set on this github repo (fixed now).

Can you elaborate?  I'm not a huge python expert.

geeksville, 2020-06-18:
oh I see - it was the lack of the auto markdown parser in pip 19.  Ok - I guess it is all sorted.  I'll add a note saying pip 20 is required.

commit 7457d334 (2020-06-18): add note saying pip 20 is required.  Fix #1
closed 2020-06-18 by geeksville