meshtastic/python #2: Likely bug mentioned on forum

geeksville, 2020-06-17:


geeksville, 2020-06-17:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/3-device-mesh-error-on-device-connected-via-usb/450/2

nicetechguy, 2020-06-17:
Error at end of log appears when second node is powered on in range.
Just a guess..Could this be due to each device being upgraded/reflashed one at a time from 0.7.4 to 0.7.5 and not being recongised correctly even though it had the same ID?

[log-error-meshtastic-202006180839.txt](https://github.com/meshtastic/Meshtastic-python/files/4795388/log-error-meshtastic-202006180839.txt)

geeksville, 2020-06-17:
oh thanks!  No it is a bug in the python library.  It is the libaries responsibility to update the nodedb when new nodes show up and it isn't doing that I'll fix now.

geeksville, 2020-06-18:
Thank you for the report @nicetechguy.  Fixed in 0.7.5 just released to pypi.

closed 2020-06-18 by geeksville

nicetechguy, 2020-06-18:
Thank you for the prompt assist!

Excited to see how this project progresses going forwards - hoping it will be the best fit for our use case :)

nicetechguy, 2020-06-18:
updated just now and encountering another error:

[log-error-meshtastic-202006181030.txt](https://github.com/meshtastic/Meshtastic-python/files/4795647/log-error-meshtastic-202006181030.txt)

nicetechguy, 2020-06-18:
Looks good now on 0.7.6 - thanks :)

commit 8ff5fc0e (2020-06-18): Fix #2 - update nodedb when we receive user or permission
commit cb4ec626 (2020-06-18): oops sorry @nicetechguy, fix #2. in scripting languages without static
validation, gotta check all execution paths before checking in ;-)