meshtastic/python #6: meshtastic-python and Flask interaction

whi-tw, 2020-08-06:
Hi,

I'm trying to make a dashboard for use on scout camps, using meshtastic as the transport method.

I've gotten it up and running and have two nodes communicating, and can happily send messages between them using python. However, as soon as I start flask up within the same python app, I start seeing weird errors:

```
WARNING:werkzeug:WebSocket transport not available. Install eventlet or gevent and gevent-websocket for improved performance.
 * Serving Flask app "app" (lazy loading)
 * Environment: development
 * Debug mode: on
INFO:werkzeug: * Running on http://127.0.0.1:5000/ (Press CTRL+C to quit)
INFO:werkzeug: * Restarting with stat
Traceback (most recent call last):
  File "/home/tnwhitwell/workspace/scouts/meshtastic-scouts/.direnv/python-3.8.1/lib/python3.8/site-packages/serial/serialposix.py", line 310, in _reconfigure_port
    fcntl.flock(self.fd, fcntl.LOCK_EX | fcntl.LOCK_NB)
BlockingIOError: [Errno 11] Resource temporarily unavailable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/tnwhitwell/workspace/scouts/meshtastic-scouts/app.py", line 141, in <module>
    meshDevice = ScoutMesh()
  File "/home/tnwhitwell/workspace/scouts/meshtastic-scouts/my_meshtastic.py", line 11, in __init__
    self.interface = meshtastic.StreamInterface()
  File "/home/tnwhitwell/workspace/scouts/meshtastic-scouts/.direnv/python-3.8.1/lib/python3.8/site-packages/meshtastic/__init__.py", line 391, in __init__
    self.stream = serial.Serial(
  File "/home/tnwhitwell/workspace/scouts/meshtastic-scouts/.direnv/python-3.8.1/lib/python3.8/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/home/tnwhitwell/workspace/scouts/meshtastic-scouts/.direnv/python-3.8.1/lib/python3.8/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/home/tnwhitwell/workspace/scouts/meshtastic-scouts/.direnv/python-3.8.1/lib/python3.8/site-packages/serial/serialposix.py", line 312, in _reconfigure_port
    raise SerialException(msg.errno, "Could not exclusively lock port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 11] Could not exclusively lock port /dev/ttyUSB0: [Errno 11] Resource temporarily unavailable
```
code:
```
    meshDevice = ScoutMesh()
    meshDevice.subscribe(extract_mesh_message, "meshtastic.receive")
    # meshDevice.interface.close()
    time.sleep(5)
    socketio.run(app)
```

(ScoutMesh is a simple wrapper class that provides a `subscribe` method, without importing pubsub all over the place)

If I close the interface, the server starts up, if I don't run the flask server, I don't see any issues.

I don't suppose you'd know off-hand where this issue may come from?

geeksville, 2020-08-06:
hmm - is the connection to the meshtastic api happening before you call flask?  or after you start flask?

my guess is that flask is using a different userid that is not in the dialout group (and therfore when we go to use the serial port we are not allowed access).  If that theory is correct, then you could add flask to dialout and I bet it would work.

No big deal, but in the future (so that it can help other devs), for API questions can you post to meshtastic.discourse.group instead?  To keep github tidy I'm going to close this issue - but definitely start a thread if you are still seeing problems.

closed 2020-08-06 by geeksville

whi-tw, 2020-08-07:
Not meaning to necro this, just wanted to make sure if someone else finds this they found the solution too:

This is the same issue as on https://github.com/pallets/flask/issues/2951, and the same fixes apply.

geeksville, 2020-08-07:
ooh that's a great find!