meshtastic/python #7: "meshtastic.receive.data" throws KeyError if receiving data via interface.sendData()

whi-tw, 2020-08-08:
Hi,

When sending data in the form `interface.sendData(b'some-bytes'), the following is thrown:
```
ERROR:root:Error handling FromRadio, possibly corrupted? 'typ'
Traceback (most recent call last):
  File "/home/tnwhitwell/workspace/scouts/meshtastic-scouts/.direnv/python-3.8.1/lib/python3.8/site-packages/meshtastic/__init__.py", line 460, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/home/tnwhitwell/workspace/scouts/meshtastic-scouts/.direnv/python-3.8.1/lib/python3.8/site-packages/meshtastic/__init__.py", line 226, in _handleFromRadio
    self._handlePacketFromRadio(fromRadio.packet)
  File "/home/tnwhitwell/workspace/scouts/meshtastic-scouts/.direnv/python-3.8.1/lib/python3.8/site-packages/meshtastic/__init__.py", line 312, in _handlePacketFromRadio
    if asDict["decoded"]["data"]["typ"] == "CLEAR_TEXT":
KeyError: 'typ'
```

I've resolved this locally by changing
https://github.com/meshtastic/Meshtastic-python/blob/9da57e2d12ee440bd291d85f2ae64521529444ec/meshtastic/__init__.py#L310-L312

to:

```python
            if "typ" in asDict["decoded"]["data"]:
                if asDict["decoded"]["data"]["typ"] == "CLEAR_TEXT":
                    asDict["decoded"]["data"]["text"] = meshPacket.decoded.data.payload.decode(
                        "utf-8")
```

I doubt this is actually the right solution, as the datatype is being set:
https://github.com/meshtastic/Meshtastic-python/blob/9da57e2d12ee440bd291d85f2ae64521529444ec/meshtastic/__init__.py#L114

Running it through the debugger, it appears `mesh_pb2.Data.OPAQUE` is resolving to `False`, and results in it not being transmitted? maybe?

geeksville, 2020-08-10:
thanks - I'll fixe this soon.

geeksville, 2020-08-12:
Thanks!  Your fix is actually almost right.  You found a problem that OPAQUE is the default protobuf typ, and protobuf default values don't get populated.  I've tweaked the receiving code to add OPAQUE if no typ was specified.

commit 0c9c8bca (2020-08-12): Fix #7: Properly decode OPAQUE binary packets...
and add binary sending to the integration tests
closed 2020-08-12 by geeksville