meshtastic/python #18: Device crashes when interfaced with via python script

mc-hamster, 2020-09-16:
Device: tbeam
Firmware version: 1.0.0 and the version in my Branch.
Python API version: 1.0.8
Desktop OS version: Ubuntu 64bit 20.04.1 & MacOS 10.15.6

This happens in every combination between 1.0.0 and the WiFi / HTTP version in my branch and using Ubuntu and MacOS.

Conditions

1) meshtastic --device /dev/cu.SLAB_USBtoUART --info

2) meshtastic --device /dev/cu.SLAB_USBtoUART --setstr wifi_ssid "asdf" --debug

This will return a lot of stats, preference and channel settings. As soon as the script returns to the prompt, the device will "freeze". After freezing, none of the buttons will work. If on a screen that has dynamic elements, those elements will update.

**Output from first run:**

jmcasler@Mac-mini python scripts % /Users/jmcasler/Library/Python/3.7/bin/meshtastic --device /dev/cu.SLAB_USBtoUART --setstr wifi_ssid "" --debug
DEBUG:root:Connecting to /dev/cu.SLAB_USBtoUART
DEBUG:root:Sending: want_config_id: 42

Trigger powerFSM 9
Trigger powerFSM 11
Transition powerFSM transition=serial API, from=ON to=SERIAL
Shutdown BT: 236468 heap size
DEBUG:root:Received: {'myInfo': {'myNodeNum': 862626501, 'hasGps': True, 'numChannels': 13, 'region': '1.0-US', 'hwModel': 'tbeam', 'firmwareVersion': '1.0.0', 'packetIdBits': 32, 'currentPacketId': 1577150038, 'nodeNumBits': 32, 'messageTimeoutMsec': 300000, 'minAppVersion': 172}}
DEBUG:root:Received: {'radio': {'preferences': {'positionBroadcastSecs': 900, 'sendOwnerInterval': 4, 'waitBluetoothSecs': 120, 'screenOnSecs': 300, 'phoneTimeoutSecs': 900, 'phoneSdsTimeoutSec': 7200, 'meshSdsTimeoutSecs': 7200, 'sdsSecs': 31536000, 'lsSecs': 3600}, 'channelSettings': {'modemConfig': 'Bw125Cr48Sf4096', 'psk': '1PG7OiApB1nwvP+rz05pvw==', 'name': 'Default'}}}
DEBUG:root:Received: {'nodeInfo': {'num': 862626501, 'user': {'id': '!c44f336aa2c5', 'longName': 'Unknown a2c5', 'shortName': '?C5', 'macaddr': 'xE8zaqLF'}, 'position': {}}}
DEBUG:root:Received: {'configCompleteId': 42}
Connected to radio
Setting preference wifi_ssid to 
Writing modified preferences to device
DEBUG:root:Sending: set_radio {
  preferences {
    position_broadcast_secs: 900
    send_owner_interval: 4
    wait_bluetooth_secs: 120
    screen_on_secs: 300
    phone_timeout_secs: 900
    phone_sds_timeout_sec: 7200
    mesh_sds_timeout_secs: 7200
    sds_secs: 31536000
    ls_secs: 3600
  }
  channel_settings {
    modem_config: Bw125Cr48Sf4096
    psk: "\324\361\273: )\007Y\360\274\377\253\317Ni\277"
    name: "Default"
  }
}

DEBUG:root:Closing serial stream
DEBUG:root:reader is exiting

**On second run, the python script will not return any data except:**

DEBUG:root:Connecting to /dev/cu.SLAB_USBtoUART
DEBUG:root:Sending: want_config_id: 42

Any string set by "--setstr" will not be saved into the device.

**Note**

About 1/10 of the time, the when device hangs, even the reset button doesn't work and the device needs to be power cycled.

geeksville, 2020-09-16:
ooh this one was super interesting.  I was able to repro from your great description.  The problem was that leaving rts low was disabling the reset button on the board.  So I think the only fix needed was to raise rts high before we close the port.

geeksville, 2020-09-16:
@mc-hamster I investigated on the web and I think it is even better to apply RTS control only on OS-X (so that on other platforms the reset button still works while connected via the API).  Can you try this new change (in version 1.0.10 on pypi) and confirm it still doesn't reboot on you?

Supposedly 'Darwin' is the string I should use to detect OS-X.

mc-hamster, 2020-09-17:
I can write wifi_ssid from Ubuntu and it fails on MacOS.

I'll switch to Linux to use the API for the time being.

mc-hamster, 2020-09-17:
Issue closed

commit 8c86a49b (2020-09-16): fix #18 reset button was disabled

ooh this one was super interesting.  I was able to repro from your great description.  The problem was that leaving rts low was disabling the reset button on the board.  So I think the only fix needed was to raise rts high before we close the port.

Meshtastic-python$ bin/run.sh --setstr wifi_ssid "asdf" --debug
rm: cannot remove 'log_*': No such file or directory
DEBUG:root:Connecting to /dev/ttyUSB0
DEBUG:root:Sending: want_config_id: 42

Trigger powerFSM 9
DEBUG:root:Received: {'myInfo': {'myNodeNum': 2883444536, 'hasGps': True, 'numChannels': 13, 'region' [... cut]
closed 2020-09-17 by mc-hamster