meshtastic/python #26: 1.1.0 cmd tool crashes

claesg, 2020-10-07:
There seems to be a problem when starting the  command line tool.

```
$ uname -a
Linux big-ubuntu 5.4.0-48-generic #52-Ubuntu SMP Thu Sep 10 10:58:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

$ pip3 install --upgrade meshtastic
Requirement already up-to-date: meshtastic in /home/claes/.local/lib/python3.8/site-packages (1.1.0)
Requirement already satisfied, skipping upgrade: pypubsub>=4.0.3 in /home/claes/.local/lib/python3.8/site-packages (from meshtastic) (4.0.3)
Requirement already satisfied, skipping upgrade: pygatt>=4.0.5 in /home/claes/.local/lib/python3.8/site-packages (from meshtastic) (4.0.5)
Requirement already satisfied, skipping upgrade: pyqrcode>=1.2.1 in /home/claes/.local/lib/python3.8/site-packages (from meshtastic) (1.2.1)
Requirement already satisfied, skipping upgrade: pexpect>=4.6.0 in /usr/lib/python3/dist-packages (from meshtastic) (4.6.0)
Requirement already satisfied, skipping upgrade: protobuf>=3.6.1 in /usr/lib/python3/dist-packages (from meshtastic) (3.6.1)
Requirement already satisfied, skipping upgrade: dotmap>=1.3.14 in /home/claes/.local/lib/python3.8/site-packages (from meshtastic) (1.3.14)
Requirement already satisfied, skipping upgrade: pyserial>=3.4 in /home/claes/.local/lib/python3.8/site-packages (from meshtastic) (3.4)
Requirement already satisfied, skipping upgrade: enum-compat in /home/claes/.local/lib/python3.8/site-packages (from pygatt>=4.0.5->meshtastic) (0.0.3)

$ meshtastic
Traceback (most recent call last):
  File "/home/claes/.local/bin/meshtastic", line 5, in <module>
    from meshtastic.__main__ import main
  File "/home/claes/.local/lib/python3.8/site-packages/meshtastic/__init__.py", line 69, in <module>
    from . import mesh_pb2
  File "/home/claes/.local/lib/python3.8/site-packages/meshtastic/mesh_pb2.py", line 22, in <module>
    create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
$ 
```
I get the same fault when importing meshtastic in a python3 prompt.
Device has just been upgraded to 1.1.1

claesg, 2020-10-07:
I managed to find a workaround (using google)
A newer protobuf solves the problem.
```
$ pip3 install --upgrade protobuf
Collecting protobuf
  Downloading protobuf-3.13.0-cp38-cp38-manylinux1_x86_64.whl (1.3 MB)
     |████████████████████████████████| 1.3 MB 8.0 MB/s 
Requirement already satisfied, skipping upgrade: setuptools in /usr/lib/python3/dist-packages (from protobuf) (45.2.0)
Requirement already satisfied, skipping upgrade: six>=1.9 in /usr/lib/python3/dist-packages (from protobuf) (1.14.0)
Installing collected packages: protobuf
Successfully installed protobuf-3.13.0

$ meshtastic 
from=LS to=DARK
Allowing GPS lock
Setting bluetooth enable=1
Pre BT: 211280 heap size
Starting bluetooth
Trigger powerFSM 9
Transition powerFSM transition=Contact from phone, from=DARK to=DARK
Trigger powerFSM 11
```

geeksville, 2020-10-08:
oh thanks - I'll update the dependency to say we need 3.13 of protobuf lib.

commit 7d269b74 (2020-10-08): fix #26 thanks @claesg - use newer protobuf lib
closed 2020-10-08 by geeksville