meshtastic/python #35: Use the TAP/TUN driver to somehow provide IP connectivity over meshtastic

geeksville, 2020-11-25:
Misc notes on this idea:

* A good gist on tap/tun from python (see comments re python3): https://gist.github.com/glacjay/585369
* Even better a lib that works on linux and windows: https://pypi.org/project/python-pytuntap/

geeksville, 2020-12-24:
Works!  (Though I'll need to make a stress test and allow forwarding over multiple hops)  I've now successfully done pings and even SSH over the lora link)

```
meshtastic --tunnel 
rm: cannot remove 'log_*': No such file or directory
Connected to radio
INFO:root:Starting IP to mesh tunnel (you must be root for this *pre-alpha* feature to work).  Mesh members:
INFO:root:Node !2462abf84098 has IP address 10.115.64.152
INFO:root:Node !2462abdddf38 has IP address 10.115.223.56
INFO:root:Node !fd1004ec5d56 has IP address 10.115.93.86
INFO:root:Node !246f28979058 has IP address 10.115.144.88
```

Useful snippets for testing:
```
# sudo bin/run.sh --port /dev/ttyUSB0 --setch-shortfast
# sudo bin/run.sh --port /dev/ttyUSB0 --tunnel --debug
# ssh -Y root@192.168.10.151 (or dietpi), default password p
# ncat -e /bin/cat -k -u -l 1235 
# ncat -u 10.115.64.152 1235
# ping -c 1 -W 20 10.115.64.152
```

geeksville, 2020-12-24:
reopening to add the following optimization:

btw: we currently send the entire IP packet for every packet. An easy optimization would be to strip out the 20 byte header (because it is mostly redundant with our mesh info), which would probably cut ping times in half.

(we should use two different APP ids - one for the current naive encoding and one for the 'headerless' encoding.  The receiver can regen a suitable fake header)

geeksville, 2020-12-24:
TODO: make tunnel addons default, until then users should

pip install --upgrade 'meshtastic[tunnel]'

http://peak.telecommunity.com/DevCenter/setuptools#declaring-extras-optional-features-with-their-own-dependencies

yhaenggi, 2020-12-28:
@geeksville When i was working (quite some time ago) with building tun IP tunnels over slow links, i added compression of the packets (back then with lzo). Compress every packet and check if its smaller afterwards (sometimes its bigger, if its not compressable). On the receiving side i checked for the magic bytes of the compression and decompressed it, before putting it on the interface. This increased throughput without a significant overhead. It also means less air-time, so its good for everyone.
Another thing is the tunnel queue length. Interfaces can buffer packets and send them out later. This way you could wait a few milliseconds and then compress all of those packets into a smaller one.

Thanks for this, i'll will try it out.

kareiva, 2021-01-02:
We have launched an IRC server on a Raspberry Pi connected to an EU433 TTGO T-Beam and are actually chatting over it right now. As suggested, MTU had to go down to 200 for this to work, but it sort of works and, combined with proper IRC ping/pong timeout settings, it's pretty cool and usable. Is anyone working on an option to provide the tunnel device MTU value directly from CLI?

https://twitter.com/kareiva/status/1345492737110507521

geeksville, 2021-01-03:
hmm - I thought my python code was setting the mtu already (i.e. I'm surprised that you needed to set it also).  so I've clearly got something wrong there.

kareiva, 2021-01-03:
> hmm - I thought my python code was setting the mtu already (i.e. I'm surprised that you needed to set it also). so I've clearly got something wrong there.

Try this: https://github.com/meshtastic/Meshtastic-python/pull/50

kareiva, 2021-01-20:
Thanks for the merge, @geeksville !

BTW, we have tested out possible MTU values and it seems that 200 is the safest option. I have had troubles connecting over TCP with MTU of 240, like it was suggested in the code comments. But 200 is a quite stable value, even over a long radio distance.

closed 2021-02-06 by geeksville

marek22k, 2023-11-16:
Hello, maybe a stupid question: But what happens if two devices would theoretically get the same IP address?

milkpirate, 2024-01-30:
Hey, I really would like to get this working, but on the one hand I am having problem to piece the parts together on the other, `meshtastic --tunnel` does not seem to be recognized:
```cmd
C:\> meshtastic --tunnel
usage: meshtastic [-h] [--configure CONFIGURE] [--export-config] [--port PORT] [--host HOST] [--seriallog SERIALLOG] [--info]
...
                  [--pos-fields [POS_FIELDS ...]] [--debug] [--test] [--ble BLE] [--ble-scan] [--noproto] [--version] [--support]
meshtastic: error: unrecognized arguments: --tunnel
```
Thought there seems to exist a `mesh-tunnel` with the same set of options.

My setup looks like so:
```mermaid
graph LR
    Win[Windows Laptop]
    <--> |Bluetooth| NodeA
    <--> |LORA| NodeB
    subgraph WiFi
    NodeB
    Ubuntu
    end
    WiFi <--> Inet((Internet))
```
And I would like to give the `Windows Laptop` access to `Ubuntu` as well as the internet.

Could you tell me if its even possible? And if so how? Maybe we can write up a tutorial for others to come.

jdstroy, 2024-05-23:
@milkpirate 

Oh dear.  So the `--tunnel` option on Meshtastic Python CLI is [only available on Linux](https://github.com/meshtastic/python/blob/master/meshtastic/__main__.py#L840), because the [pytap2](https://github.com/meshtastic/python/blob/master/meshtastic/tunnel.py#L23) library used in this implementation [doesn't support Windows](https://github.com/johnthagen/pytap2/blob/master/src/pytap2/__init__.py).

There is a [C# library](https://github.com/meshtastic/c-sharp) (and CLI) for accessing Meshtastic devices that's supported; I've done some prototyping [on an unofficial fork](https://github.com/jdstroy/meshtastic-csharp) which includes support for a "tunnel" sub-command, compatible with the Python CLI.  If you intend to use it, you will need to download the [Wintun](https://www.wintun.net/) library; I've provided more extensive instructions for using Wintun with SharpTun [here](https://github.com/SharpTun/SharpTun), which I've used in my implementation of the `tunnel` sub-command on my fork of the C# CLI.  SharpTun requires wintun.dll, and the tunnel command requires SharpTun.

It's probably worth noting up-front that the Meshtastic firmware does not (currently?) use the IP_TUNNEL_APP messages at all; you will need to run something like the Python CLI with `--tunnel` for your Windows Laptop to reach anything through `NodeB` over IPv4.  As far as I know, none of the API clients (Python, C# library) support Meshtastic over BT as a transport, especially on Windows; you will need to use the USB port (Meshtastic over serial), or Wi-Fi (Meshtastic over TCP).  You will also need to select a faster radio preset on both nodes, like `ShortFast`, to use the tunnel feature (IPv4 over Meshtastic), as both Windows and Linux respond poorly to high latency links (such as Meshtastic in `LongFast` or `LongSlow` presets).

commit b86a4194 (2020-12-24): IP tunnel works!  fix #35