meshtastic/python #39: Can't send text with meshtastic.py --sendtext

timgunter, 2020-12-12:
When I run "meshtastic --sendtext some_message", onConnected() closes the interface before the message is sent. Since onConnected is called from a thread managed by pubsub, i'm not sure what the best way to resolve this is. Do we need a way to check if the message has been sent? Or to optionally block until it has been sent? Do we need to add an isBusy threading.Event to MeshInterface?

timgunter, 2020-12-13:
Perhaps the best thing to do is just set closeNow to False

geeksville, 2020-12-13:
ooh thanks I think this is a problem with how sent bytes are being treated when closing the serial port.  It seems like they might be getting dropped when the port is closed (even though we previously called serial.flush() to wait for them getting sent.  I think this might mean the problem you saw is a dup of #41 

@timgunter can I ask if you are using windows? (I'm having a hard time reproing this on linux)

geeksville, 2020-12-13:
I just created a nasty possible fix (see my commit in #41) - would you mind trying it on your machine?

geeksville, 2020-12-13:
closing as a dup of #41

closed 2020-12-13 by geeksville