meshtastic/python #41: preference changes not saving in 1.1.8

geeksville, 2020-12-13:
reported by @dinthenorth in the alpha tester thread.  alas, I haven't yet been able to reproduce this.  My run seems successful:

```
~/development/meshtastic/Meshtastic-python$ meshtastic --set position_broadcast_secs 60 --set wait_bluetooth_secs 28800
rm: cannot remove 'log_*': No such file or directory
Node changed: {'num': 2885173400, 'user': {'id': '!2462abf84098', 'longName': 'Unknown 4098', 'shortName': '?98', 'macaddr': 'JGKr+ECY'}, 'position': {'batteryLevel': 100, 'time': 1605398970}}
Connected to radio
Setting position_broadcast_secs to 60
Setting wait_bluetooth_secs to 28800
Writing modified preferences to device
~/development/meshtastic/Meshtastic-python$ meshtastic --info
Node changed: {'num': 2885173400, 'user': {'id': '!2462abf84098', 'longName': 'Unknown 4098', 'shortName': '?98', 'macaddr': 'JGKr+ECY'}, 'position': {'batteryLevel': 100, 'time': 1605398970}}
Connected to radio
my_node_num: 2885173400
has_gps: true
num_channels: 10
region: "1.0"
hw_model: "tbeam"
firmware_version: "1.1.8"
packet_id_bits: 32
current_packet_id: 1401156767
node_num_bits: 32
message_timeout_msec: 300000
min_app_version: 172

preferences {
  position_broadcast_secs: 60
  wait_bluetooth_secs: 28800
  ls_secs: 300
  region: TW
}
channel_settings {
  modem_config: Bw125Cr48Sf4096
  psk: "\324\361\273: )\007Y\360\274\377\253\317Ni\277"
  name: "Default"
}

Channel URL https://www.meshtastic.org/c/#GAMiENTxuzogKQdZ8Lz_q89Oab8qB0RlZmF1bHQ=
Nodes in mesh:
{'num': 2885173400, 'user': {'id': '!2462abf84098', 'longName': 'Unknown 4098', 'shortName': '?98', 'macaddr': 'JGKr+ECY'}, 'position': {'batteryLevel': 100, 'time': 1605398970}}
```

To folks who see this problem a couple of questions:

1. can you try "pip3 install --upgrade meshtastic" to make sure you have the latest python tool?
2. What OS is your desktop computer?

geeksville, 2020-12-13:
@timgunter if my commit above doesn't fix the bug you found, would you mind try calling time.sleep(0.2) in StreamInterface.__disconnected() before calling close() on the port.  I bet that will fix this problem (by waiting for any last sent characters to be delivered from the PC to the device before the port is closed)

(My theory is that pyserial flush() is not working correctly on Windows - but I don't have a windows machine to test with)

timgunter, 2020-12-13:
Just tried the latest commit, and I'm still having the same issue. A .2 second sleep in StreamInterface._disconnected() doesn't solve the issue. When I sleep for about 10 seconds, the message does get through.

I noticed that on raspbian, when the python app disconnects it doesn't reboot the device. However, on my Windows box, the device is rebooted both when the python api connects and when it disconnects. Perhaps this is the issue?

geeksville, 2020-12-13:
> I noticed that on raspbian, when the python app disconnects it doesn't reboot the device. However, on my Windows box, the device is rebooted both when the python api connects and when it disconnects. Perhaps this is the issue?

OOOH!  Thats a good clue.  Yeah - that's not supposed to happen and doesn't happen on my linux desktop.  What model of lora device are you using?

geeksville, 2020-12-13:
@timgunter Could you try removing this conditional and applying the same fix that worked for OS-X?

in __init__.py
        # OS-X seems to have a bug in its serial driver.  It ignores that we asked for no RTSCTS
        # control and will always drive RTS either high or low (rather than letting the CP102 leave
        # it as an open-collector floating pin).  Since it is going to drive it anyways we want to make
        # sure it is driven low, so that the TBEAM won't reset
        if platform.system() == 'Darwin':
            self.stream.rts = False

geeksville, 2020-12-13:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/firmware-complete-with-wifi-enabled/1858/20

timgunter, 2020-12-13:
I have experienced the device reboot when connecting from Windows with all the devices I have:
TTGO T-Beam v1.1 w/ SX1276
TTGO T-Beam v1.1 w/ SX1262
Heltec WiFi Lora32 V2

The RTS fix also works on Windows. Now the device doesn't reset, and the messages get through.

geeksville, 2020-12-13:
yay thanks for that info!  I'll change the polarity of that fix to apply on all platforms that are not linux

timgunter, 2020-12-14:
Just to make things more complicated, on my Windows system, setting rts to True in SerialInterface._disconnected() results in a device reset. The value of rts is "False" prior to setting it to True here. If I set rts to False in __init__(), and do nothing in_disconnected(), the message gets through, the device does not reset, and the reset button on the device still works. If I re-invoke meshtastic.py to send another message just after the first one, the value of rts upon entering __init__() is True.

geeksville, 2020-12-16:
@timgunter ooh that's super helpful.  Would it be possible for you to convert that fix into a PR?  I don't have a windows machine to test on currently and it sounds like you've found the perfect solution (and a lot of people are having this problem)

geeksville, 2020-12-17:
1.1.25 released with this fix.  Anyone who is in this bug thread, would you mind downloading and confirm it works for you?  (ideally someone with Windows and someone else with OS-X - I just tested linux)

timgunter, 2020-12-17:
Works on my Windows box. Thanks!

commit 86005c0d (2020-12-13): 1.1.23 fix #41 pyserial also drives usb control signals wrong on windows
closed 2020-12-19 by geeksville