meshtastic/python #44: text message subscription

FooBarDude, 2020-12-14:
Hi
Maybe I'm getting it wrong, but to get text messages sent by app I have to subscribe to "meshtastic.receive.data" if I subscribe to "meshtastic.receive.text" I'm not getting any messages.

timgunter, 2020-12-14:
I think this is a consequence of #42 . You will need to decode the payload data to get a proper string. You can also try pull request #43 and see if that fixes the issue for you.

geeksville, 2020-12-15:
@FooBarDude I changed how text messages are handled (sorry about that) because I wanted to make them uniform with all other data packet types.  I intended for meshtastic.receive.text should still be working for you.  I'll look at @timgunter's fix #43 and merge a variant of it now.

geeksville, 2020-12-15:
fix release to pypi thanks @timgunter and sorry about the breakage @FooBarDude ;-)

closed 2020-12-15 by geeksville

FooBarDude, 2020-12-19:
Just now had the time to test it.
Works.
Thanks a lot!

commit 862f6606 (2020-12-15): A quick fix #44 mostly by @timgunter.  thanks!