meshtastic/python #46: Attatched node not sending

FooBarDude, 2020-12-26:
I did some testing today and had a strange issue.
I used a program on my computer to captures the GPS positions, and send replies if it receives a message, so I can check if the LoRa is still in range.
This worked fine first, but after a short while, the replies sent from the computer didn't get through to the mobile node.
The computer got messages from my mobile node, generated replies and sent them to the attached node (they were showing on the display), but somehow they didn't transmit to my mobile node. Only after resetting the Node, I could get it working again.
My guess is that the attached node somehow gets stuck and doesn't send out messages it received over USB, but I don't know how to verify if that's acutely the case.

I used:
AppVersion 1.1.22
Firmware 1.1.23
Meshtastic-python 1.1.25

geeksville, 2021-01-08:
hmm - do you have a link to the program I could look at?

FooBarDude, 2021-01-11:
It's still quite rough, but I since I wanted to put it on github at some point anyway I just made the initial commit:
https://github.com/FooBarDude/meshtastic-tracker

geeksville, 2021-02-15:
sorry - I forgot about this bug.  Is it working now?

crossan007, 2021-02-18:
I saw this too on 1.2x.

I don't think I've seen it on 1.4x.

geeksville, 2021-03-10:
sounds quiet so I'm assuming fixed.  Please reopen if still a problem with latest.

closed 2021-03-10 by geeksville