meshtastic/python #48: TCP API connection closing ends all connections from that IP

ScriptBlock, 2020-12-30:
Scenario:
Server 1 launches messagelistener.py
   messagelistener.py begins TCP connection to meshtastic API.  
   Subscribes to meshtastic.receive.  
   Begins receiving message.  
   Begins looping, waiting for messages - delivering downstream once received.  
This continues to run indefinitely.

Server 1 launches info.py
   info.py begins TCP connection to meshtastic API.  
   Subscribes to meshtastic.connection.established. 
   Once connection is made, queries interface.myInfo 
info.py closes the connection.

Once info.py completes, messagelistener.py no longer receives any messages, but also throws no errors.  I can't tell if this is a problem with the API making multiple TCP connections or if the device is closing all API connections for a given IP.

I realize that this may not be the desired implementation of the API (i.e. - multiple TCP API requests from the same machine).  In which case I can modify things to use a single thread/script, but it's less than ideal in this case.

Thanks!

ScriptBlock, 2020-12-30:
i didn't see anything obvious or apparent in the serial logs or any helpful error messages or behavior from the python script.

geeksville, 2020-12-31:
hmm - (just to clarify) you have two TCP clients talking to one device simultaneously?  Alas, no that's not supported right now.  It probably won't be too hard to fix.  Alas my queue is kinda busy for the next few days but I can take a look later next week.