meshtastic/python #52: Problem sending consecutive messages

jinx17, 2021-01-11:
Hi,
I did a simple python script to send a message.
Device is connected to linux computer. 
When I send first message it goes through to all nodes in the mesh network. I see it on all OLED displays.
Then, when I start script again, only the device attached to the computer shows message on a display, but all other nodes don't get this message.
If I wait around 3 minutes, and then start script again, message is transferred to all nodes in mesh netwok (I see message appear on all node displays).

I think there is a problem with module not releasing resources (serialport) after it is finnished, allthough I use interface.close ...

I run my script like this: 
`/root/mesh_sendtext1.py SOMEARGUMENT`
and display should show text like: Message: SOMEARGUMENT

my script:
`#!/usr/bin/env python3
import meshtastic, sys, atexit, os
n = len(sys.argv)
interface = meshtastic.SerialInterface(devPath="/dev/ttyUSB0")
interface.sendText("Message:  " + sys.argv[1])
print("\nStevilo argumentov: ",sys.argv[1])
atexit.register(print,"Program exited successfully!")
interface.close()
del interface
os._exit(1)
`
Best regards,
Jinx

geeksville, 2021-03-10:
oh @jnx17 - I somehow missed the notify for this bug.  Things have improved a lot since Jan would you mind testing with latest and see if you still see this problem?

geeksville, 2021-05-24:
no reply from reporter closing.

closed 2021-05-24 by geeksville