meshtastic/python #86: Tag the source

fabaff, 2021-04-26:
It would be very helpful if you could tag releases. This would enable distributions who want to run your tests to fetch the package from GitHub instead of Pypi, where the tests are excluded.

Thanks

geeksville, 2021-05-24:
ok will do

mkinney, 2021-11-22:
Not sure if it helps, but this is a *release.sh* script I used in another project: (with the project specific values changed to XXX)

```
#!/bin/sh

set -e

echo "Change XXX/__init__.py version."
echo "Update the CHANGELOG.md file."
echo "Update the README.md file, if applicable."
echo "Commit those changes before running this script."
echo "Create/merge the pull request."

echo "***Be sure you are on master branch.***"

VERSION=`python setup.py --version`

echo "# Releasing XXX v$VERSION..."

echo "# Running tests..."
pytest

git push

echo "# Git tag & push..."
git tag -a "v$VERSION" -m "v$VERSION"
git push --tags

echo "# Upload to pypi..."
# Clear build & dist
rm -rf build/* dist/*
# Build source and wheel packages
python setup.py sdist bdist_wheel
# Upload w/Twine
twine upload dist/*

echo "# All done!"
```

mkinney, 2021-12-30:
@mc-hamster This is being done now, right? Can we close this issue?

mkinney, 2022-01-06:
Closing issue.

closed 2022-01-06 by mkinney