meshtastic/python #90: Errno 104 - TCP problems with meshtastic 1.2.33 and 1.2.30 firmware

pwilen, 2021-05-11:
when executing "meshtastic --host 192.168.68.xxx --info" I get:
ERROR:root:Unexpected OSError, terminating meshtastic reader... [Errno 104] Connection reset by peer

I have tested with a T-beam and a Heltec. The units  are connected to my WiFi-router and I can access their (limited) web interface 192.168.68.xxx/static/index.html

Tested on raspbian and OS X

jfirwin, 2021-05-21:
On Slack, @thebentern and I looked into this about the same time this issue was posted.

To summarize our findings:

| Meshtastic-python | Meshtastic-device | `--host` successful |
| :-----------------:  | :-----------------: | :-------------------: |
| 1.2.23 | 1.2.11 | TRUE |
| 1.2.33 | 1.2.11 | TRUE |
| 1.2.33 | 1.2.16 | FALSE |
| 1.2.33 | 1.2.30 | FALSE |

I couldn't test one of the firmware release between 1.2.11 and 1.2.16 because of an error in the flashing script I believe.

@thebentern found some [changes](https://github.com/meshtastic/Meshtastic-device/compare/1.2.13...1.2.16#diff-443f04602bc3e75b1d48daef0ac158122d2391fe83712a7046e699dbe7e2ae2c) to the device code StreamAPI.cpp

geeksville, 2021-05-23:
looking at this now.  thanks for the report.

geeksville, 2021-05-24:
This turns out to be a device bug.  Moving there...