meshtastic/python #95: CLI --ch-set psk random not working

jfirwin, 2021-05-27:
```
~ $ meshtastic --ch-set psk random
Connected to radio
Aborting due to: name 'genPSK256' is not defined
```

I believe the issue has to do with [these lines](https://github.com/meshtastic/Meshtastic-python/blob/037d29e2007efae971d3c05ca721d11f3b6d9976/meshtastic/__main__.py#L75-L76) based on the error message.

geeksville, 2021-05-27:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/encryption-clarification/3425/4

jfirwin, 2021-05-27:
I'm on macOS, I'll confirm this is the case with windows as well. On macOS I saw the following in terminal based off of [this line](https://github.com/meshtastic/Meshtastic-python/blob/037d29e2007efae971d3c05ca721d11f3b6d9976/meshtastic/__main__.py#L68). 
```
~ $ python3
Python 3.9.0 (v3.9.0:9cf6752276, Oct  5 2020, 11:29:23) 
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.urandom(32)
b'[byte information]'
>>> 
```

```
~ $ python3
Python 3.9.2 (default, Feb 19 2021, 17:09:53) 
[Clang 12.0.0 (clang-1200.0.32.29)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.urandom(32)
b'[byte information]'
>>>
```

jfirwin, 2021-05-27:
Confirmed on Windows 10 running gitbash that same error
```
Aborting due to: name 'genPSK256' is not defined
```

geeksville, 2021-05-28:
1.2.36 just pushed to pypi!

commit b734b9f0 (2021-05-28): fix #95, somehow a S got inserted ;-) (thanks @jfirwin)
closed 2021-05-28 by geeksville