meshtastic/python #98: Receive text API call?

sokoow, 2021-06-25:
Hi All,

I trawled your objects and docs seeking for non blocking or at least blocking receive call in this API - but it's missing. Any plans on adding it ?

geeksville, 2021-07-15:
Received messages (in both text and binary) are asynchronously delivered as publishes.  So you just subscribe to meshtastic.receive.data.<portnum> or just meshtastic.receive (if you want to receive all messages).  Or if you just want text meshtastic.receive.text.  per https://github.com/meshtastic/Meshtastic-python/blob/master/meshtastic/__init__.py#L24

Example usage here:
https://github.com/meshtastic/Meshtastic-python/blob/master/meshtastic/test.py#L148

Does that help?

geeksville, 2021-07-15:
btw @sokoow sorry for the late reply - just noticed this one...

sokoow, 2021-07-16:
thanks @geeksville

closed 2021-07-16 by sokoow