meshtastic/python #103: --ch-set or similar should support meshtastic URLs

bensonk, 2021-08-23:
Channel URLs are an easy way to share meshtastic network details. It would be incredibly valuable for the channel set functionality to accept a URL, to make it easy for a CLI-controlled node to join a network shared from an android app.

mkinney, 2022-01-06:
There is the `--seturl` capability in the command line utility as well as the ability to create a QR code from a setting with the `--qr` option.

Did you want to be able scan the qr code in the Android app? (if so, then maybe we need to move this issue over to that repo)

mkinney, 2022-01-27:
I think the answer above resolves the issue. If not, feel free to re-open this issue or open a new issue.

closed 2022-01-27 by mkinney