meshtastic/python #104: meshtastic --tunnel -> sh: 1: ifconfig: not found

KiwiHC16, 2021-09-04:
Hi,

I'm trying to setip the tunnel for test purposes. I got an error about ifconfig.

I did the test on a VM Debian 10 and on a VM Ubuntu.

Followed the steps from pages:

- https://github.com/meshtastic/Meshtastic-python
- https://meshtastic.discourse.group/t/ip-over-meshtastic-kinda-implemented/1775/5

After a french install of ubuntu here are the command I used as root:

```
$apt install python3-pip
$pip3 install --upgrade pytap2
$pip3 install --upgrade meshtastic

$ls -l /dev/ttyUSB*
crw-rw---- 1 root dialout 188, 0 Sep  4 03:00 /dev/ttyUSB0

$meshtastic -h
(I get in return all help text)

$meshtastic --tunnel
Connected to radio
INFO:root:Starting IP to mesh tunnel (you must be root for this *pre-alpha* feature to work).  Mesh members:
INFO:root:Node !f244c500 has IP address 10.115.197.0
INFO:root:Node !f244c480 has IP address 10.115.196.128
sh: 1: ifconfig: not found
Aborting due to: 
```
 
After  putting in debug I can identify the issue at line https://github.com/meshtastic/Meshtastic-python/blob/8380e57fa060b4cd5a46cfce7ec4fcc99019c3ed/meshtastic/tunnel.py#L102

If I do a small test.py

File
```
from pytap2 import TapDevice
tun = TapDevice(name="mesh")
tun.up()
tun.ifconfig(address="192.168.4.2", netmask="255.255.255.0", mtu=200)
```

Test
```
root@abeilleDev:~# python3 test.py 
sh: 1: ifconfig: not found
Traceback (most recent call last):
  File "test.py", line 3, in <module>
    tun.up()
  File "/usr/local/lib/python3.7/dist-packages/pytap2/__init__.py", line 213, in up
    raise IfconfigError()
pytap2.IfconfigError
```

Identical error message.

Looks like there is an issue with the lib.

Any clue from anybody ?

geeksville, 2021-09-04:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/meshtastic-tunnel-sh-1-ifconfig-not-found/3818/1

KiwiHC16, 2021-09-04:
https://github.com/johnthagen/pytap2/issues/8

KiwiHC16, 2021-09-04:
https://linuxconfig.org/how-to-install-missing-ifconfig-command-on-debian-linux

```
apt-get install net-tools
```

this issue looks solve but I fall into an other one:
```
root@host:~# meshtastic --tunnel
Connected to radio
INFO:root:Starting IP to mesh tunnel (you must be root for this *pre-alpha* feature to work).  Mesh members:
INFO:root:Node !f244c480 has IP address 10.115.196.128
INFO:root:Node !f244c500 has IP address 10.115.197.0
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.7/dist-packages/meshtastic/tunnel.py", line 170, in __tunReader
    p = tap.read()
  File "/usr/local/lib/python3.7/dist-packages/pytap2/__init__.py", line 144, in read
    return os.read(self._fd, num_bytes)
OSError: [Errno 9] Bad file descriptor
```

@geeksville which Linux distrubion do you use ?

KiwiHC16, 2021-09-04:
I'm not lucky ;-(
Now with a Fedora bistro I'm getting:
```
[root@localhost ~]# meshtastic --tunnel
Connected to radio
INFO:root:Starting IP to mesh tunnel (you must be root for this *pre-alpha* feature to work).  Mesh members:
INFO:root:Node !f244c480 has IP address 10.115.196.128
INFO:root:Node !f244c500 has IP address 10.115.197.0
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib64/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib64/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.8/site-packages/meshtastic/tunnel.py", line 170, in __tunReader
    p = tap.read()
  File "/usr/local/lib/python3.8/site-packages/pytap2/__init__.py", line 144, in read
    return os.read(self._fd, num_bytes)
OSError: [Errno 9] Bad file descriptor

```

KiwiHC16, 2021-09-04:
```
meshtastic --reboot 
```
works. I see the device rebooting.

KiwiHC16, 2021-09-04:
@geeksville On which firmware version tunnel was tested ?
(I have FW 1.2.45.b674054)

KiwiHC16, 2021-09-04:
Same behavior with firmware-tbeam-1.2.5.bin

KiwiHC16, 2021-09-04:
@#96 ?

markbirss, 2021-09-04:
@KiwiHC16 

I also cant get tunnel working

KiwiHC16, 2021-09-04:
```
cd /usr/local/lib/python3.7/dist-packages/meshtastic
vi tunnel.py
```

Comment Thread and call __tunReader() around line 104:
```
        logging.debug(f"starting TUN reader, our IP address is {myAddr}")
        self.__tunReader()
        # self._rxThread = threading.Thread( target=self.__tunReader, args=(), daemon=True)
        # self._rxThread.start()
```

Launch meshtastic --tunnel

mesh0 interface is up
```
ifconfig
mesh0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 200
        inet 10.115.197.0  netmask 255.255.0.0  destination 10.115.197.0
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1  bytes 48 (48.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Ping is ok:
```
# ping 10.115.197.0
PING 10.115.197.0 (10.115.197.0) 56(84) bytes of data.
64 bytes from 10.115.197.0: icmp_seq=1 ttl=64 time=0.022 ms
64 bytes from 10.115.197.0: icmp_seq=2 ttl=64 time=0.034 ms
64 bytes from 10.115.197.0: icmp_seq=3 ttl=64 time=0.033 ms
```

KiwiHC16, 2021-09-05:
Modification done into tunnel.py

Around line 100:
```
        from pytap2 import TapDevice
        self.tun = TapDevice(name="mesh")
        self.tun.up()
        self.tun.ifconfig(address=myAddr, netmask=netmask, mtu=200)
        logging.info(f"starting TUN reader, our IP address is {myAddr}")

        self._rxThread = threading.Thread( target=self.__tunReader, args=(), daemon=True)
        self._rxThread.start()
```

With modification:
```
        from pytap2 import TapDevice
        self.tun = TapDevice(name="mesh")
        self.tun.up()
        self.tun.ifconfig(address=myAddr, netmask=netmask, mtu=200)
        logging.info(f"starting TUN reader, our IP address is {myAddr}")
        while True:
            p = self.tun.read()
            logging.info(f"IP packet received on TUN interface, type={type(p)}")
            destAddr = p[16:20]
            if not self._shouldFilterPacket(p):
                self.sendPacket(destAddr, p)
        # self._rxThread = threading.Thread( target=self.__tunReader, args=(ttaapp,), daemon=True)
        # self._rxThread.start()
```

Then everything work. I can ping, ssh and http a web page.

Seams to confirm an issue in relation with the Thread.

KiwiHC16, 2021-09-05:
So if we shift some lines in the code then it fix the issue:
In __init__ function we keep only:
```
        self._rxThread = threading.Thread( target=self.__tunReader, args=(ttaapp,), daemon=True)
        self._rxThread.start()
```

and we move the lines to __tunReader function which give:
```
    def __tunReader(self):
        logging.info("creating TUN device with MTU=200")
        # FIXME - figure out real max MTU, it should be 240 - the overhead bytes for SubPacket and Data
        from pytap2 import TapDevice
        tun = TapDevice(name="mesh")
        tun.up()
        tun.ifconfig(address=myAddr, netmask=netmask, mtu=200)
        logging.info(f"starting TUN reader, our IP address is {myAddr}")
        logging.info("KiwiHC16: TUN reader running")
        while True:
            p = tun.read()
            logging.info(f"IP packet received on TUN interface, type={type(p)}")
            destAddr = p[16:20]

            if not self._shouldFilterPacket(p):
                self.sendPacket(destAddr, p)
```

jdstroy, 2021-09-05:
Take a look at #106; see if that works for you.

markbirss, 2021-09-05:
@jdstroy  

 #106;, fixed the issue for me, thank you

https://github.com/meshtastic/Meshtastic-python/pull/106

KiwiHC16, 2021-09-05:
You PR looks so simple !
I'll test it as soon as I can.

KiwiHC16, 2021-09-05:
I just tested between:
RPI3 - USB - TBeam - Lora - TBeam - USB - VM Debian

Pings work in both directions.

Thanks a lot.

@geeksville please inject the PR in the Trunk.

I'm closing the issue.

closed 2021-09-05 by KiwiHC16

marek22k, 2023-12-05:
On Linux distributions ifconfig is no longer up to date and ip should be used. The only legitimate purpose for ifconfig would be on BSD systems.