meshtastic/python #105: Remotely administering your node

KiwiHC16, 2021-09-04:
As per the documentation page: 
https://meshtastic.org/docs/software/device/device-remote-admin#creating-the-admin-channel

Admin channel created on both equipment.

Then try to change the remote owner:

```
www-data@hosts:$ meshtastic --dest \!f244c480 --set-owner "Im Remote"
Connected to radio
Setting device owner to Im Remote
INFO:root:Requesting preferences from remote node (this could take a while)
ERROR:root:Error while handling message from radio 'admin'
Traceback (most recent call last):
  File "/var/www/.local/lib/python3.7/site-packages/meshtastic/__init__.py", line 873, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/var/www/.local/lib/python3.7/site-packages/meshtastic/__init__.py", line 554, in _handleFromRadio
    self._handlePacketFromRadio(fromRadio.packet)
  File "/var/www/.local/lib/python3.7/site-packages/meshtastic/__init__.py", line 692, in _handlePacketFromRadio
    handler.callback(asDict)
  File "/var/www/.local/lib/python3.7/site-packages/meshtastic/node.py", line 291, in onResponse
    self.radioConfig = p["decoded"]["admin"]["raw"].get_radio_response
KeyError: 'admin'
Aborting due to: Timed out waiting for node config
```

KiwiHC16, 2021-09-04:
Second exemple fail in the same way:

```
meshtastic --dest \!28979058 --set ls_secs 301 --info
````

markbirss, 2021-09-04:
@KiwiHC16 

At first I though that the Bw500Cr45Sf128 (shortfast) was the issue
https://meshtastic.discourse.group/t/1-2-45-firmware-ready-for-brave-alpha-testers/3785/10?u=mark.birss

but here is what i did to get it working

`meshtastic --port /dev/ttyUSB0 --ch-index 1 --ch-del
meshtastic --port /dev/ttyUSB0 --ch-add admin
meshtastic --port /dev/ttyUSB0 --seturl https://www.meshtastic.org/d/#ChUiAQEqC1Blbmd1aW5GYXJtVdkHAAAKKSIguYkh_NzLo_gUALEuy4WyMo-acOPItOgX93JqRYesHfMqBWFkbWlu`

`meshtastic --port /dev/ttyUSB0 --dest \!2897868c --info
Connected to radio

INFO:root:Requesting preferences from remote node (this could take a while)
INFO:root:Requesting channel 0 info from remote node (this could take a while)
INFO:root:Requesting channel 1 info from remote node (this could take a while)
INFO:root:Requesting channel 2 info from remote node (this could take a while)
Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "region": "EU865" }

Channels:
  PRIMARY psk=default { "psk": "AQ==", "name": "PenguinFarm", "id": 2009 }
  SECONDARY psk=secret { "psk": "uYkh/NzLo/gUALEuy4WyMo+acOPItOgX93JqRYesHfM=", "name": "admin" }

Primary channel URL: https://www.meshtastic.org/d/#ChUiAQEqC1Blbmd1aW5GYXJtVdkHAAA
Complete URL (includes all channels): https://www.meshtastic.org/d/#ChUiAQEqC1Blbmd1aW5GYXJtVdkHAAAKKSIguYkh_NzLo_gUALEuy4WyMo-acOPItOgX93JqRYesHfMqBWFkbWlu

`

KiwiHC16, 2021-09-04:
I tried with "Short range (but fast)" and "Medium range (but fast)" and I get the same error.
With "Very long range (but slow)" I get a timeout.

I tried to delete one admin channel and re-create it in one side and still the same issue.

Will delete all and try again.

KiwiHC16, 2021-09-04:
Got it to work.
I have 2 TBeam. Lets call them master and slave.
I notice that he "master" was flashing at 1hz but with a quick flash while the "slave" was also at 1Hz but with a flash as long as the dark period. Both Red and Bleu LEDs flashing together.
From master I asked in debug mode:
```
meshtastic --port /dev/ttyUSB0 --dest \!f244c480 --info --debug
```

And it works !

Now I cando the same request without out debug and Master as Slave flash quick at 1 Hz.

With
```
meshtastic --port /dev/ttyUSB0 --dest \!f244c480 --info 
```

From master to interrogate slave, I get the info from slave.

KiwiHC16, 2021-09-04:
If I setup with your URL and reset both equipments then it works. If I use my URL and reset both equipments it doesn't work.

KiwiHC16, 2021-09-04:
With your URL:
```
meshtastic --port /dev/ttyUSB0 --seturl https://www.meshtastic.org/d/#ChUiAQEqC1Blbmd1aW5GYXJtVdkHAAAKKSIguYkh_NzLo_gUALEuy4WyMo-acOPItOgX93JqRYesHfMqBWFkbWlu
```

I get:
```
Channels:
  PRIMARY psk=default { "psk": "AQ==", "name": "PenguinFarm", "id": 2009 }
  SECONDARY psk=secret { "psk": "uYkh/NzLo/gUALEuy4WyMo+acOPItOgX93JqRYesHfM=", "name": "admin" }

Primary channel URL: https://www.meshtastic.org/d/#ChUiAQEqC1Blbmd1aW5GYXJtVdkHAAA
Complete URL (includes all channels): https://www.meshtastic.org/d/#ChUiAQEqC1Blbmd1aW5GYXJtVdkHAAAKKSIguYkh_NzLo_gUALEuy4WyMo-acOPItOgX93JqRYesHfMqBWFkbWlu
```

With my URL:
```
```

I get:
```
Channels:
  PRIMARY psk=default { "modemConfig": "Bw500Cr45Sf128", "psk": "AQ==" }
  SECONDARY psk=secret { "psk": "I97thsduDgpDVciKiOt9TcZL3+LijKj2LTfm879rXp0=", "name": "admin" }

Primary channel URL: https://www.meshtastic.org/d/#CgUYASIBAQ
Complete URL (includes all channels): https://www.meshtastic.org/d/#CgUYASIBAQopIiAj3u2Gx24OCkNVyIqI631Nxkvf4uKMqPYtN-bzv2tenSoFYWRtaW4
```

KiwiHC16, 2021-09-04:
I looked a bit more to your topic: https://meshtastic.discourse.group/t/1-2-45-firmware-ready-for-brave-alpha-testers/3785

And I used my URL and change only the modem_config on both sides:
``` 
meshtastic --ch-set modem_config Bw125Cr45Sf128  --port /dev/ttyUSB0
```

Reset both side and it works.

KiwiHC16, 2021-09-04:
**Bw500Cr45Sf128 (Fast) -> doesn't work**
Bw125Cr45Sf128 (Medium) -> works
Bw31_25Cr48Sf512 (LongAlt) -> works
Bw125Cr48Sf4096 (LongSlow)-> works

KiwiHC16, 2021-09-04:
So I see 2 issues to solve:
- admin doesn't work in Bw500Cr45Sf128 mode
- meshtastic doesn't print modemconfig if not in Bw500Cr45Sf128 mode

geeksville, 2021-09-04:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/1-2-45-firmware-ready-for-brave-alpha-testers/3785/12

KiwiHC16, 2021-09-04:
@geeksville yes this is the same conclusion. I added a comment to link the 2 topics.

joshpirihi, 2022-01-31:
I'm getting this error too:

```
 joshuapirihi@Joshuas-MacBook-Pro-3  ~  meshtastic --host 10.147.253.136 --dest \!f24750a4 --info
Connected to radio

Requesting preferences from remote node.
Be sure:
 1. There is a SECONDARY channel named 'admin'.
 2. The '--seturl' was used to configure.
 3. All devices have the same modem config. (i.e., '--ch-longfast')
 4. All devices have been rebooted after all of the above. (optional, but recommended)
Note: This could take a while (it requests remote channel configs, then writes config)
Requesting channel 0 info from remote node (this could take a while)
Requesting channel 1 info from remote node (this could take a while)
ERROR file:stream_interface.py __reader line:165 Error while handling message from radio 'admin'
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/meshtastic/stream_interface.py", line 163, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/usr/local/lib/python3.9/site-packages/meshtastic/mesh_interface.py", line 535, in _handleFromRadio
    self._handlePacketFromRadio(fromRadio.packet)
  File "/usr/local/lib/python3.9/site-packages/meshtastic/mesh_interface.py", line 685, in _handlePacketFromRadio
    handler.callback(asDict)
  File "/usr/local/lib/python3.9/site-packages/meshtastic/node.py", line 322, in onResponseRequestChannel
    c = p["decoded"]["admin"]["raw"].get_channel_response
KeyError: 'admin'

```

mcumings, 2022-02-01:
I'm also getting this failure mode. I originally tried the 1.2.52 firmware but upon failure also tried bumping to the 1.2.53 firmware with the same results.  I'm using `Bw125Cr48Sf4096` (LongSlow) and am not seeing the success mentioned by KiwiHC16 above.

Here are my node infos:

```
Connected to radio

Owner: Mojo-8380 (838)
My info: { "myNodeNum": 4145973296, "hasGps": true, "numBands": 13, "firmwareVersion": "1.2.53.19c1f9f", "rebootCount": 7, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 13.799999, "airUtilTx": 1.1956389 }
Nodes in mesh:  {'num': 4145973296, 'user': {'id': '!f71e8830', 'longName': 'Mojo-8380', 'shortName': '838', 'macaddr': '30:c6:f7:1e:88:30', 'hwModel': 'TBEAM'}, 'position': {'batteryLevel': 100}, 'lastHeard': 1643684420}  {'num': 4145975016, 'user': {'id': '!f71e8ee8', 'longName': 'Mojo-8ee8', 'shortName': '8EE', 'macaddr': '30:c6:f7:1e:8e:e8', 'hwModel': 'TBEAM'}, 'position': {'batteryLevel': 100, 'time': 1643560320}, 'lastHeard': 1643684403, 'snr': 12.5}  {'num': 4145975392, 'user': {'id': '!f71e9060', 'longName': 'Mojo-9060', 'shortName': '906', 'macaddr': '30:c6:f7:1e:90:60', 'hwModel': 'TBEAM'}, 'position': {'time': 1643560476}, 'lastHeard': 1643684265, 'snr': 10.5}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "region": "US", "positionFlags": 99 }

Channels:
  PRIMARY psk=secret { "modemConfig": "Bw125Cr48Sf4096", "psk": "DvF56GZ7dg9uRwNqPvs0KccSfPN0/kSHyUdni4Vr0qo=" }
  SECONDARY psk=secret { "psk": "fh0QumogCISQ5++z7H4+FE2A2sb1LnJxNmBlmijx1XA=", "name": "admin", "id": 1 }
  SECONDARY psk=secret { "psk": "1CWOLK7+hjgYFklGUvX1+KJwkgxGWbaxuvFpQrzd5ww=", "name": "family", "id": 2 }

Primary channel URL: https://www.meshtastic.org/d/#CiQYAyIgDvF56GZ7dg9uRwNqPvs0KccSfPN0_kSHyUdni4Vr0qo
Complete URL (includes all channels): https://www.meshtastic.org/d/#CiQYAyIgDvF56GZ7dg9uRwNqPvs0KccSfPN0_kSHyUdni4Vr0qoKLiIgfh0QumogCISQ5--z7H4-FE2A2sb1LnJxNmBlmijx1XAqBWFkbWluVQEAAAAKLyIg1CWOLK7-hjgYFklGUvX1-KJwkgxGWbaxuvFpQrzd5wwqBmZhbWlseVUCAAAA
```
and
```
Connected to radio

Owner: Mojo-8ee8 (8EE)
My info: { "myNodeNum": 4145975016, "hasGps": true, "numBands": 13, "firmwareVersion": "1.2.53.19c1f9f", "rebootCount": 5, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 27.599998, "airUtilTx": 1.2245278 }
Nodes in mesh:  {'num': 4145975016, 'user': {'id': '!f71e8ee8', 'longName': 'Mojo-8ee8', 'shortName': '8EE', 'macaddr': '30:c6:f7:1e:8e:e8', 'hwModel': 'TBEAM'}, 'position': {'batteryLevel': 100}, 'lastHeard': 1643684416}  {'num': 4145973296, 'user': {'id': '!f71e8830', 'longName': 'Mojo-8380', 'shortName': '838', 'macaddr': '30:c6:f7:1e:88:30', 'hwModel': 'TBEAM'}, 'position': {'time': 1643684421}, 'lastHeard': None, 'snr': 8.5, 'lastReceived': {'from': 4145973296, 'to': 4294967295, 'id': 1041452968, 'rxSnr': 8.5, 'hopLimit': 2, 'fromId': '!f71e8830', 'toId': '^all'}, 'hopLimit': 2}  {'num': 4145975392, 'user': {'id': '!f71e9060', 'longName': 'Mojo-9060', 'shortName': '906', 'macaddr': '30:c6:f7:1e:90:60', 'hwModel': 'TBEAM'}, 'position': {'time': 1643560739}, 'lastHeard': 1643560743, 'snr': -12.5}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "region": "US", "rangeTestPluginEnabled": true, "rangeTestPluginSender": 60, "positionFlags": 99 }

Channels:
  PRIMARY psk=secret { "modemConfig": "Bw125Cr48Sf4096", "psk": "DvF56GZ7dg9uRwNqPvs0KccSfPN0/kSHyUdni4Vr0qo=" }
  SECONDARY psk=secret { "psk": "fh0QumogCISQ5++z7H4+FE2A2sb1LnJxNmBlmijx1XA=", "name": "admin", "id": 1 }
  SECONDARY psk=secret { "psk": "1CWOLK7+hjgYFklGUvX1+KJwkgxGWbaxuvFpQrzd5ww=", "name": "family", "id": 2 }

Primary channel URL: https://www.meshtastic.org/d/#CiQYAyIgDvF56GZ7dg9uRwNqPvs0KccSfPN0_kSHyUdni4Vr0qo
Complete URL (includes all channels): https://www.meshtastic.org/d/#CiQYAyIgDvF56GZ7dg9uRwNqPvs0KccSfPN0_kSHyUdni4Vr0qoKLiIgfh0QumogCISQ5--z7H4-FE2A2sb1LnJxNmBlmijx1XAqBWFkbWluVQEAAAAKLyIg1CWOLK7-hjgYFklGUvX1-KJwkgxGWbaxuvFpQrzd5wwqBmZhbWlseVUCAAAA
```
and the `--support` output (for both):
```
If having issues with meshtastic cli or python library
or wish to make feature requests, visit:
https://github.com/meshtastic/Meshtastic-python/issues
When adding an issue, be sure to include the following info:
 System: Darwin
   Platform: macOS-10.15.7-x86_64-i386-64bit
   Release: 19.6.0
   Machine: x86_64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.2.78
 Executable: /usr/local/bin/meshtastic
 Python: 3.9.9 CPython Clang 12.0.0 (clang-1200.0.32.29)

Please add the output from the command: meshtastic --info
```

Note that one of these nodes has the range test plugin enabled and the other one does register every sequence number on its display so they appear to otherwise be working correctly.

rolouis, 2022-03-15:
Hi, looks like the issue isn't resolved yet.
Does anyone know a workaround?

yhaenggi, 2022-03-26:
Same here, another gentle bump.

KiwiHC16, 2022-03-29:
Some interesting findings today, en mode debug:

The code is looking for admin:
```
c = p["decoded"]["admin"]["raw"].get_channel_response
```

But in the message decode we don't have admin:
```
DEBUG file:node.py onResponseRequestChannel line:489 onResponseRequestChannel() p:{'from': 4064593024, 'to': 4064593152, 'channel': 1, 'decoded': {'portnum': 'ROUTING_APP', 'payload': b'\x18\x00', 'requestId': 554894865, 'routing': {'errorReason': 'NONE', 'raw': error_reason: NONE
}}, 'id': 606829195, 'rxTime': 1648582505, 'rxSnr': 13.0, 'raw': from: 4064593024
to: 4064593152
channel: 1
decoded {
  portnum: ROUTING_APP
  payload: "\030\000"
  request_id: 554894865
}
id: 606829195
rx_time: 1648582505
rx_snr: 13.0
, 'fromId': '!f244c480', 'toId': '!f244c500'}
ERROR file:stream_interface.py __reader line:171 Error while handling message from radio 'admin'
Traceback (most recent call last):
  File "/home/ben/.local/lib/python3.9/site-packages/meshtastic/stream_interface.py", line 169, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/home/ben/.local/lib/python3.9/site-packages/meshtastic/mesh_interface.py", line 536, in _handleFromRadio
    self._handlePacketFromRadio(fromRadio.packet)
  File "/home/ben/.local/lib/python3.9/site-packages/meshtastic/mesh_interface.py", line 688, in _handlePacketFromRadio
    handler.callback(asDict)
  File "/home/ben/.local/lib/python3.9/site-packages/meshtastic/node.py", line 490, in onResponseRequestChannel
    c = p["decoded"]["admin"]["raw"].get_channel_response
KeyError: 'admin'
```

Is the system trying to decode wrong message ....

KiwiHC16, 2022-03-29:
```
meshtastic --get is_router
Connected to radio
is_router: False
Completed getting preferences
```

KiwiHC16, 2022-03-30:
As we can see in the message above, I got a "random" ROUTING_APP message which has the request_id.

This is not the answer message expected for this request_id.

```
decoded {
  portnum: ROUTING_APP
  payload: "\030\000"
  request_id: 554894865
}
```

When we don't have this "random" message, everything works well, when we got it it crash the code.

I made a modification in my code to manage such situation. Looks like it works well now.

I will push a "Merge Request" to have a code more robust.

At the same time it would be good to understand why we got this message.

A bug in the FW of the meshtastic ?

KiwiHC16, 2022-03-31:
If pull request #310 approved then we can close this issue.

closed 2022-07-02 by garthvh