meshtastic/python #146: Max retransmit error is causing exception as not handled

OverByThere, 2021-12-07:
When attempting to get information on a node using `showInfo()` out of range, it returns the following error after a bit of time:
```
ERROR:root:Error while handling message from radio 'admin'
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/meshtastic/__init__.py", line 885, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/usr/local/lib/python3.7/site-packages/meshtastic/__init__.py", line 566, in _handleFromRadio
    self._handlePacketFromRadio(fromRadio.packet)
  File "/usr/local/lib/python3.7/site-packages/meshtastic/__init__.py", line 704, in _handlePacketFromRadio
    handler.callback(asDict)
  File "/usr/local/lib/python3.7/site-packages/meshtastic/node.py", line 287, in onResponse
    self.radioConfig = p["decoded"]["admin"]["raw"].get_radio_response
KeyError: 'admin'
```

I believe this is because in node.py nothing handles the error in the response:
https://github.com/meshtastic/Meshtastic-python/blob/068b902f12b077daa87cf02cdc876a2e921ff9be/meshtastic/node.py#L270

My crude update to at least report the error and carry on is as follows:
```
def onResponse(p):
    """A closure to handle the response packet"""
    errorFound = False
    if 'routing' in p["decoded"]:
        if 'errorReason' in p["decoded"]["routing"]:
            errorFound = True
            print(f'Error on response: {p["decoded"]["routing"]["errorReason"]}')
    if errorFound is False:
        self.radioConfig = p["decoded"]["admin"]["raw"].get_radio_response
        logging.debug("Received radio config, now fetching channels...")
        self._timeout.reset()  # We made foreward progress
        self._requestChannel(0)  # now start fetching channels
```

This would then result with the following error output:
`Error on response: MAX_RETRANSMIT`

I cannot propose this as a PR or a full solution as I know its terrible code, but thought I should raise.

mkinney, 2021-12-12:
I wasn't sure how to replicate the issue.

So, I followed the steps from https://meshtastic.org/docs/software/device/device-remote-admin

one device1
`meshtastic --set factory_reset true`
reboot device
`meshtastic --ch-add admin`
`meshtastic --set is_always_powered true` (optional)
`meshtastic --info` (to get the url)
on a 2nd device:
`meshtastic --set factory_reset true`
reboot device
`meshtastic --set is_always_powered true` (optional)
` meshtastic --seturl "xxx"` (where xxx was the url from the prior device)
I plugged the 2nd device into just power.
I plugged the 1st device into usb.

I was able to get the error when I ran:
`meshtastic --set-owner 24 --dest \!28b54624`

The before error/exception was the same (or very similar to the one above)
I made the suggested code changes and this is what I see now:

```
% meshtastic --nodes
Connected to radio
╒═════╤══════════════╤═══════╤═══════════╤════════════╤═════════════╤════════════╤═══════════╤═════════╤═════════════════════╤═══════════════╕
│   N │ User         │ AKA   │ ID        │ Latitude   │ Longitude   │ Altitude   │ Battery   │ SNR     │ LastHeard           │ Since         │
╞═════╪══════════════╪═══════╪═══════════╪════════════╪═════════════╪════════════╪═══════════╪═════════╪═════════════════════╪═══════════════╡
│   1 │ cc           │ cc    │ !28af67cc │ N/A        │ N/A         │ N/A        │ 39.00%    │ N/A     │ 2021-12-11 17:43:58 │ just now      │
├─────┼──────────────┼───────┼───────────┼────────────┼─────────────┼────────────┼───────────┼─────────┼─────────────────────┼───────────────┤
│   2 │ Unknown 4624 │ ?24   │ !28b54624 │ N/A        │ N/A         │ N/A        │ 59.00%    │ 5.00 dB │ 2021-12-11 17:36:27 │ 7 minutes ago │
╘═════╧══════════════╧═══════╧═══════════╧════════════╧═════════════╧════════════╧═══════════╧═════════╧═════════════════════╧═══════════════╛
(venv) sweet Meshtastic-python % meshtastic --set-owner 24 --dest \!28b54624
Connected to radio
Setting device owner to 24
INFO:root:Requesting preferences from remote node (this could take a while)
Error on response: MAX_RETRANSMIT
Error: Timed out waiting for node config
```

So, it is not a fix per se, but it definitely makes the error(s) look better.

mkinney, 2021-12-12:
I think this may help the error, but not fix the underlying issue. Ok to close this issue? @OverByThere

mkinney, 2021-12-12:
see also https://github.com/meshtastic/Meshtastic-python/issues/105

OverByThere, 2021-12-12:
Happy to close, hope the underlying issue in #105 gets resolved soon to mitigate this issue :)

closed 2021-12-12 by OverByThere