meshtastic/python #159: 1.2.46.dce2fe4 alpha - Every other message via Meshtastic python fails

1n5aN1aC, 2021-12-12:
**Describe the bug**
I was beginning testing of 1.2.46.dce2fe4 alpha and I usually do initial testing via the meshtastic python app. (`pip3 install meshtastic`)
However, exactly every other command fails with the message `Exception: Timed out waiting for connection completion`

I did not experience this issue when I was on 1.2.45 beta

**Device info:**
 - Device model: [TLORA-v1]
 - firmwareVersion: [1.2.46.dce2fe4]
 - Meshtastic tool: [1.2.40]

I have another of these devices I will test on and report back eventually.
I also have a t-beam device available I'll test on eventually.

**Examples:**

```
PS C:\Users\Snip\Desktop\meshtastic> meshtastic --port COM7 --nodes
Connected to radio
╒═════╤══════════════╤═══════╤═══════════╤═══════╤═════════════════════╤════════════════╕
│   N │ User         │ AKA   │ ID        │ SNR   │ LastHeard           │ Since          │
╞═════╪══════════════╪═══════╪═══════════╪═══════╪═════════════════════╪════════════════╡
│   1 │ Unknown 4f34 │ ?34   │ !11114f34 │ N/A   │ 2021-10-26 12:11:23 │ 38 seconds ago │
╘═════╧══════════════╧═══════╧═══════════╧═══════╧═════════════════════╧════════════════╛
PS C:\Users\Snip\Desktop\meshtastic> meshtastic --port COM7 --set-owner "TLORA No Tape"
Traceback (most recent call last):
  File "C:\Program Files\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python39\Scripts\meshtastic.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__main__.py", line 663, in main
    common()
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__main__.py", line 508, in common
    client = SerialInterface(
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 936, in __init__
    StreamInterface.__init__(
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 773, in __init__
    self.connect()
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 795, in connect
    self._waitConnected()
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 424, in _waitConnected
    raise Exception("Timed out waiting for connection completion")
Exception: Timed out waiting for connection completion
PS C:\Users\Snip\Desktop\meshtastic> meshtastic --port COM7 --set-owner "test"
Connected to radio
Setting device owner to test
PS C:\Users\Snip\Desktop\meshtastic> meshtastic --port COM7 --nodes
Traceback (most recent call last):
  File "C:\Program Files\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python39\Scripts\meshtastic.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__main__.py", line 663, in main
    common()
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__main__.py", line 508, in common
    client = SerialInterface(
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 936, in __init__
    StreamInterface.__init__(
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 773, in __init__
    self.connect()
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 795, in connect
    self._waitConnected()
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 424, in _waitConnected
    raise Exception("Timed out waiting for connection completion")
Exception: Timed out waiting for connection completion
PS C:\Users\Snip\Desktop\meshtastic> meshtastic --port COM7 --nodes
Connected to radio
╒═════╤══════════════╤═══════╤═══════════╤═══════╤═════════════════════╤════════════════╕
│   N │ User         │ AKA   │ ID        │ SNR   │ LastHeard           │ Since          │
╞═════╪══════════════╪═══════╪═══════════╪═══════╪═════════════════════╪════════════════╡
│   1 │ Unknown 4f34 │ ?34   │ !11114f34 │ N/A   │ 2021-10-26 12:12:30 │ 39 seconds ago │
╘═════╧══════════════╧═══════╧═══════════╧═══════╧═════════════════════╧════════════════╛
PS C:\Users\Snip\Desktop\meshtastic> meshtastic --port COM7 --nodes
Traceback (most recent call last):
  File "C:\Program Files\Python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program Files\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Program Files\Python39\Scripts\meshtastic.exe\__main__.py", line 7, in <module>
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__main__.py", line 663, in main
    common()
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__main__.py", line 508, in common
    client = SerialInterface(
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 936, in __init__
    StreamInterface.__init__(
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 773, in __init__
    self.connect()
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 795, in connect
    self._waitConnected()
  File "C:\Program Files\Python39\lib\site-packages\meshtastic\__init__.py", line 424, in _waitConnected
    raise Exception("Timed out waiting for connection completion")
Exception: Timed out waiting for connection completion
PS C:\Users\Snip\Desktop\meshtastic> meshtastic --port COM7 --nodes
Connected to radio
╒═════╤══════════════╤═══════╤═══════════╤═══════╤═════════════════════╤════════════════╕
│   N │ User         │ AKA   │ ID        │ SNR   │ LastHeard           │ Since          │
╞═════╪══════════════╪═══════╪═══════════╪═══════╪═════════════════════╪════════════════╡
│   1 │ Unknown 4f34 │ ?34   │ !11114f34 │ N/A   │ 2021-10-26 12:12:48 │ 38 seconds ago │
╘═════╧══════════════╧═══════╧═══════════╧═══════╧═════════════════════╧════════════════╛
PS C:\Users\Snip\Desktop\meshtastic>
```

1n5aN1aC, 2021-12-24:
Just a note that this issue still happens with 1.2.49.5354c49 alpha

I'm somewhat hesitant to call it a meshtastic-python issue, it feel more like a firmware issue with this particular board, but I'm not sure...

mkinney, 2021-12-24:
Have you updated your mesthastic lib? `pip install meshtastic --upgrade`

If so, you can add a few more details like `meshtastic --support`, `meshtastic --info --nodes`

Sadly, I don't have a T-Lora device.

1n5aN1aC, 2021-12-24:
Yeah, they work OK, but the TBeams and other devices are superior in most ways.  :)

> meshtastic --support
If having issues with meshtastic cli or python library
or wish to make feature requests, visit:
https://github.com/meshtastic/Meshtastic-python/issues
When adding an issue, be sure to include the following info:
 System: Windows
   Platform: Windows-10
   Release: 10
   Machine: AMD64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.2.46
 Executable: C:\Program Files\Python39\Scripts\meshtastic
 Python: 3.9.7 CPython MSC v.1929 64 bit (AMD64)

> meshtastic --port COM7 --nodes --info
Connected to radio

Owner: TLora With Red Tape (TWR)
My info: { "myNodeNum": 123456789, "numBands": 20, "firmwareVersion": "1.2.49.5354c49", "rebootCount": 49, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8 }
Nodes in mesh:  {'num': 123456789, 'user': {'id': '!1a2b3c4d', 'longName': 'TLora With Red Tape', 'shortName': 'TWR', 'macaddr': '1a2b3c4d', 'hwModel': 'TLORA_V1'}, 'position': {}, 'lastHeard': 1640388434}  {'num': 111222333, 'user': {'id': '!11aa22bb', 'longName': 'Yellow Box', 'shortName': 'YB', 'macaddr': '11aa22bb', 'hwModel': 'TBEAM'}, 'position': {'time': 1640388335}, 'lastHeard': 1640388408, 'snr': 11.5}  {'num': 11111111, 'user': {'id': '!22223333', 'longName': 'Grey Box', 'shortName': 'GB', 'macaddr': '22223333', 'hwModel': 'TBEAM'}, 'position': {'time': 1640388400}, 'lastHeard': 1640388429, 'snr': 8.0}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "region": "US", "positionBroadcastSmart": true }

Channels:
  PRIMARY psk=default { "modemConfig": "Bw125Cr48Sf4096", "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/d/#CgUYAyIBAQ

╒═════╤═════════════════════╤═══════╤═══════════╤══════════╤═════════════════════╤════════════════╤════════════╤═════════════╤════════════╤═══════════╕
│   N │ User                │ AKA   │ ID        │ SNR      │ LastHeard           │ Since          │ Latitude   │ Longitude   │ Altitude   │ Battery   │
╞═════╪═════════════════════╪═══════╪═══════════╪══════════╪═════════════════════╪════════════════╪════════════╪═════════════╪════════════╪═══════════╡
│   1 │ TLora With Red Tape │ TWR   │ !1a2b3c4d │ N/A      │ 2021-12-24 15:27:14 │ just now       │ N/A        │ N/A         │ N/A        │ N/A       │
├─────┼─────────────────────┼───────┼───────────┼──────────┼─────────────────────┼────────────────┼────────────┼─────────────┼────────────┼───────────┤
│   2 │ Grey Box            │ GB    │ !22223333 │ 8.00 dB  │ 2021-12-24 15:27:09 │ 11 seconds ago │ N/A        │ N/A         │ N/A        │ N/A       │
├─────┼─────────────────────┼───────┼───────────┼──────────┼─────────────────────┼────────────────┼────────────┼─────────────┼────────────┼───────────┤
│   3 │ Yellow Box          │ YB    │ !11aa22bb │ 11.50 dB │ 2021-12-24 15:26:48 │ 32 seconds ago │ N/A        │ N/A         │ N/A        │ N/A       │
╘═════╧═════════════════════╧═══════╧═══════════╧══════════╧═════════════════════╧════════════════╧════════════╧═════════════╧════════════╧═══════════╛

mkinney, 2022-01-06:
There was some re-writing of serial comms recently. Can you re-test with Lora devices and see if it is any better now?

1n5aN1aC, 2022-01-13:
This issue appears to be resolved as of 1.2.50.41dcfdd beta.

I upgraded to `firmware-tlora-v1-1.2.50.41dcfdd.bin`, and no longer experience this issue.

Thanks!

closed 2022-01-13 by 1n5aN1aC