meshtastic/python #168: TCP connection problem in version 1.2.46

ANS-spb, 2021-12-17:
If I'm trying connect to radio via WiFi from python, I'm getting error:

```
    interface = meshtastic.TCPInterface(hostname='192.168.1.28')
AttributeError: module 'meshtastic' has no attribute 'TCPInterface'
```
Everything worked on the previous version 1.2.45

I'm used this method:
`interface = meshtastic.TCPInterface("192.168.1.28")`

Method from manual not working too
`interface = meshtastic.TCPInterface(hostname='192.168.1.28')`

Meshtastic-python ver. 1.2.46

mkinney, 2021-12-17:
What does the import statements look like?

ANS-spb, 2021-12-17:
> What does the import statements look like?

```
import meshtastic
import time
from pubsub import pub
```

According to the manual from https://meshtastic.org/docs/software/python/python-usage

mkinney, 2021-12-17:
Sorry, I broke that accidentally recently when adding pylint and unit testing. I’m not at a computer right now, but will be in a few hours.

For this new version of meshtastic, you’ll have to temp change imports, or your could downgrade to prior version of the library.

I can be more specific on both options after I verify them myself.

mkinney, 2021-12-21:
I don't like it, but I think we now have to add another import statement and change the call to the TCPInterface class.

This is because the TCPInterface class in prior releases was all in the __init__.py file. This class has been broken out to the tcp_interface.py file.

mkinney, 2021-12-21:
Also updated that doc: see https://github.com/meshtastic/Meshtastic/pull/148

PR #174 (2021-12-21): add examples; add Makefile
fix #168
closed 2021-12-21 by mkinney