meshtastic/python #172: CLI commands show 'id' error

jaykayenn, 2021-12-21:
All CLI commands now preceded with this error:

```
ERROR:root:Error while handling message from radio 'id'
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/stream_interface.py", line 156, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "/usr/local/lib/python3.8/dist-packages/meshtastic/mesh_interface.py", line 486, in _handleFromRadio
    self.nodes[node["user"]["id"]] = node
KeyError: 'id'
Connected to radio

Owner: TBeam 2 (T2)
My info: { "myNodeNum": 2127706616, "hasGps": true, "numBands": 10, "firmwareVersion": "1.2.49.5354c49", "rebootCount": 28, "bitrate": 95.41063, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8 }
Nodes in mesh:  {'num': 2127706616, 'user': {'id': '!7ed239f8', 'longName': 'TBeam 2', 'shortName': 'T2', 'macaddr': 'lLl+0jn4', 'hwModel': 'TBEAM'}, 'position': {'batteryLevel': 100}, 'lastHeard': 1640063460}  {'num': 2127707136, 'user': {'id': '!7ed23c00', 'longName': 'TBeam 1', 'shortName': 'T1', 'macaddr': 'lLl+0jwA', 'hwModel': 'TBEAM'}, 'position': {'batteryLevel': 93, 'time': 1639842126}, 'lastHeard': 1639842221, 'snr': 10.75}

Preferences: { "positionBroadcastSecs": 375, "sendOwnerInterval": 9, "phoneTimeoutSecs": 900, "lsSecs": 30, "region": "TW", "positionBroadcastSmart": true, "locationShare": "LocEnabled", "gpsUpdateInterval": 375, "positionFlags": 35 }

Channels:
  PRIMARY psk=default { "modemConfig": "Bw250Cr46Sf2048", "psk": "AQ==" }
```

Downgrading to 1.2.44 same error. 

 System: Linux
   Platform: Linux-5.4.0-91-generic-x86_64-with-glibc2.29
   Release: 5.4.0-91-generic
   Machine: x86_64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.2.46
 Executable: /usr/local/bin/meshtastic
 Python: 3.8.10 CPython GCC 9.3.0

jaykayenn, 2021-12-21:
[py debug.txt](https://github.com/meshtastic/Meshtastic-python/files/7749926/py.debug.txt)

mkinney, 2021-12-21:
Thanks for your help getting the debugging info. Was able to write a unit test to check for that condition and prevent the exception from showing up.

closed 2021-12-21 by mkinney