meshtastic/python #184: have the auto generated doc include private methods

mkinney, 2021-12-29:
Looks like the pdoc tool will not generate methods/variables that start with an underscore, as that is the pythonic way to specify "private".

Beings that Meshtastic-python is a dev library, that is not very helpful.

I've narrowed it down to this line:
https://github.com/mkinney/pdoc/blob/master/pdoc/__init__.py#L409

I've changed it to this:

`return not ident_name.startswith("__")`

Now our doc includes more of what we want. I need to see if I can pass a command line arg to pdoc or add a config setting or something.

mkinney, 2021-12-29:
Opened issue about this: https://github.com/pdoc3/pdoc/issues/375

SubhaNAG2001, 2025-07-12:
Based on your investigation into `pdoc`'s behavior with underscore-prefixed members, here's a comprehensive solution to customize the documentation output without modifying `pdoc`'s source code:

### 1. **Monkey-Patching Approach** (Recommended)
Create a wrapper script that modifies `pdoc`'s filtering behavior at runtime:

```python
# custom_pdoc.py
import pdoc
import sys

# Override the private member filter
def custom_is_public(ident_name: str) -> bool:
    """Custom filter that only excludes dunder methods (__internal__)"""
    return not ident_name.startswith("__")

# Monkey-patch pdoc's internal function
pdoc.extract._is_public = custom_is_public

# Import CLI after patching
from pdoc.cli import main

if __name__ == "__main__":
    sys.argv[0] = "pdoc"  # Maintain CLI appearance
    main()
```

Run with:
```bash
python custom_pdoc.py meshtastic --html --output-dir docs
```

### 2. **Subclassing Approach** (More Maintainable)
Create a custom documentation generator:

```python
# custom_docgen.py
from pdoc import doc, extract

class CustomDocGenerator(doc.Doc):
    def _is_public(self, ident_name: str) -> bool:
        return not ident_name.startswith("__")

def custom_get_doc(module) -> CustomDocGenerator:
    """Factory function for custom doc generator"""
    return CustomDocGenerator(module)

# Override the default generator
extract.get_doc = custom_get_doc

# Now run standard pdoc CLI
from pdoc.cli import main
import sys

if __name__ == "__main__":
    sys.exit(main())
```

### 3. **Template Customization**
Modify Jinja templates to include private members (preserves pdoc's filtering but shows more):

1. Clone default templates:
```bash
pdoc --html --output-dir templates meshtastic --template-dir templates
```

2. Modify `templates/module.html.jinja2`:
```jinja
{% for name, class in module.classes | dictsort %}
<div class="doc doc-class" id="{{ name }}">
    <h1>{{ name }}</h1>
    <div class="doc doc-contents">
        {# Show ALL members including private ones #}
        {% for member in class.members | sort %}
            {% include "class_member.html.jinja2" %}
        {% endfor %}
    </div>
</div>
{% endfor %}
```

3. Run with custom templates:
```bash
pdoc --html --output-dir docs meshtastic --template-dir templates
```

### 4. **pdoc Configuration Module**
Create a configuration module that patches on import:

```python
# pdoc_config.py
import pdoc.extract

def _is_public(ident_name: str) -> bool:
    return not ident_name.startswith("__")

pdoc.extract._is_public = _is_public
```

Run with:
```bash
PDOC_CONFIG_MODULE=pdoc_config pdoc meshtastic
```

### Key Notes:
1. **Dunder Method Handling**:  
   The solutions preserve filtering for *true* private methods (like `__enter__`, `__init__`) while showing "protected" members (`_internal_method`).

2. **Template Customization Benefits**:
   ```jinja
   {# Show protected members #}
   {% for member in class.members if not member.name.startswith('__') %}
   ```

3. **Combined Approach** (Recommended for Meshtastic):
   ```python
   # .meshtastic_doc.py
   import pdoc.doc
   
   class CustomDoc(pdoc.doc.Doc):
       def _is_public(self, name: str) -> bool:
           return True  # Show EVERYTHING
   
   pdoc.doc.Doc = CustomDoc
   
   if __name__ == "__main__":
       import pdoc.cli
       pdoc.cli.main()
   ```

Run with:
```bash
python .meshtastic_doc.py --html meshtastic
```

These solutions give you granular control over documentation output while keeping `pdoc` upgrades clean. The monkey-patching approach (#1) is ideal for CI/CD pipelines, while template customization (#3) provides the most design control.