meshtastic/python #185: Request persistent config to set serialUSB port

jaykayenn, 2021-12-29:
Meshtastic boards appear on /dev/ttyACM for me while my /ttyUSB ports are permanently occupies by laptop modems. This requires including --port in every CLI command. 

Request a text config.ini to persistently set the desired Meshtastic port. 

Thanks

```
DEBUG:root:Not logging serial output
DEBUG:root:ports:['/dev/ttyACM0', '/dev/ttyUSB0', '/dev/ttyUSB1', '/dev/ttyUSB2']
Warning: Multiple serial ports were detected so one serial port must be specified with the '--port'.
  Ports detected:['/dev/ttyACM0', '/dev/ttyUSB0', '/dev/ttyUSB1', '/dev/ttyUSB2']
```

mkinney, 2021-12-29:
I thought of a non-coding option.

Create a script like this:
```
meshtastic --port /dev/ttyACM0 "$@"
```

Put that script somewhere in your path, then just use that instead of using `meshtastic` directly.

jaykayenn, 2021-12-30:
> I thought of a non-coding option.
> 
> Create a script like this:
> 
> ```
> meshtastic --port /dev/ttyACM0 "$@"
> ```
> 
> Put that script somewhere in your path, then just use that instead of using `meshtastic` directly.

That's useful, thanks!

mkinney, 2021-12-30:
Great! Ok to close this issue?

jaykayenn, 2021-12-30:
Workaround with local shell script.

closed 2021-12-30 by jaykayenn