meshtastic/python #192: CLI info showing raw and decoded data together

jaykayenn, 2021-12-30:
The --info query has returned this strange output
[image]

Meshtastic-python .48
Device firmware .50

mkinney, 2021-12-30:
Hmm... I have the same setup, but not seeing the same thing.

Here's what I'm seeing:
```
(venv) sweet Meshtastic-python % meshtastic --info
Connected to radio

Owner: Unknown 8244 (?44)
My info: { "myNodeNum": 2475131460, "hasGps": true, "numBands": 13, "firmwareVersion": "1.2.50.41dcfdd", "rebootCount": 146, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true }
Nodes in mesh:  {'num': 2475131460, 'user': {'id': '!93878244', 'longName': 'Unknown 8244', 'shortName': '?44', 'macaddr': 'RBeTh4JE', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': 457314729, 'longitudeI': -1226697217, 'altitude': 60, 'time': 1640879615, 'locationSource': 'LOCSRC_GPS_INTERNAL', 'posTimestamp': 1640879610, 'altitudeHae': 40, 'altGeoidSep': -20, 'PDOP': 1030, 'fixType': 3, 'satsInView': 4, 'latitude': 45.7314729, 'longitude': -122.6697217}, 'lastHeard': 1640879615}  {'num': 2475227164, 'user': {'id': '!9388f81c', 'longName': 'Unknown f81c', 'shortName': '?1C', 'macaddr': 'RBeTiPgc', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': 457317556, 'longitudeI': -1226698700, 'altitude': 61, 'time': 1640879458, 'latitude': 45.7317556, 'longitude': -122.66986999999999}, 'lastHeard': 1640879466, 'snr': 10.5}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "positionFlags": 35 }

Channels:
  PRIMARY psk=default { "modemConfig": "Bw125Cr48Sf4096", "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/d/#CgUYAyIBAQ
```

mkinney, 2021-12-30:
Ah... I was able to reproduce it (even if temporarily).

I added wifi_ssid and wifi_password and reset the TBeam. It generated an SSL cert. 

Now, I'm seeing it:
```
(venv) sweet Meshtastic-python % meshtastic --info
Connected to radio

Owner: Unknown 8244 (?44)
My info: { "myNodeNum": 2475131460, "hasGps": true, "numBands": 13, "firmwareVersion": "1.2.50.41dcfdd", "rebootCount": 147, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 12.695 }
Nodes in mesh:  {'num': 2475131460, 'user': {'id': '!93878244', 'longName': 'Unknown 8244', 'shortName': '?44', 'macaddr': 'RBeTh4JE', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': 457318371, 'longitudeI': -1226700072, 'altitude': 67, 'time': 1640880124, 'locationSource': 'LOCSRC_GPS_INTERNAL', 'posTimestamp': 1640880121, 'altitudeHae': 46, 'altGeoidSep': -20, 'PDOP': 676, 'fixType': 3, 'satsInView': 5, 'latitude': 45.7318371, 'longitude': -122.6700072}, 'lastHeard': 1640880124}  {'num': 2475227164, 'user': {'id': '!9388f81c', 'longName': 'Unknown f81c', 'shortName': '?1C', 'macaddr': 'RBeTiPgc', 'hwModel': 'TBEAM', 'raw': id: "!9388f81c" long_name: "Unknown f81c" short_name: "?1C" macaddr: "D\027\223\210\370\034" hw_model: TBEAM }, 'position': {'latitudeI': 457317556, 'longitudeI': -1226698700, 'altitude': 61, 'time': 1640879458, 'latitude': 45.7317556, 'longitude': -122.66986999999999}, 'lastHeard': 1640880120, 'snr': 9.75, 'lastReceived': {'from': 2475227164, 'to': 2475131460, 'decoded': {'portnum': 'NODEINFO_APP', 'payload': b'\n\t!9388f81c\x12\x0cUnknown f81c\x1a\x03?1C"\x06D\x17\x93\x88\xf8\x1c0\x04', 'requestId': 1739783114, 'user': {'id': '!9388f81c', 'longName': 'Unknown f81c', 'shortName': '?1C', 'macaddr': 'RBeTiPgc', 'hwModel': 'TBEAM', 'raw': id: "!9388f81c" long_name: "Unknown f81c" short_name: "?1C" macaddr: "D\027\223\210\370\034" hw_model: TBEAM }}, 'id': 1835451332, 'rxTime': 1640880120, 'rxSnr': 9.75, 'hopLimit': 3, 'raw': from: 2475227164 to: 2475131460 decoded { portnum: NODEINFO_APP payload: "\n\t!9388f81c\022\014Unknown f81c\032\003?1C\"\006D\027\223\210\370\0340\004" request_id: 1739783114 } id: 1835451332 rx_time: 1640880120 rx_snr: 9.75 hop_limit: 3 , 'fromId': '!9388f81c', 'toId': '!93878244'}, 'hopLimit': 3}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "wifiSsid": "sofast6", "wifiPassword": "sekrit", "positionFlags": 35 }

Channels:
  PRIMARY psk=default { "modemConfig": "Bw125Cr48Sf4096", "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/d/#CgUYAyIBAQ
```

mkinney, 2021-12-30:
But, it goes away... hmm.. so not wifi related... just right after the device is restarted.

mkinney, 2021-12-30:
I think just the 'raw' field needed to be removed from that output. If you see something like that after the next release, please re-open this issue (or open a new issue).

Thanks for providing the feedback

mkinney, 2021-12-30:
We did a release today, so this fix should be in there.

mkinney, 2022-01-02:
It happened again, so re-opening issue.

mkinney, 2022-01-02:
Just pushed 1.2.51 that has some changes that might prevent this.

@jaykayenn Can you test/validate? If you don't see the issue anymore, feel free to re-close the issue.

jaykayenn, 2022-01-03:
Looks good now. Closing.

PR #194 (2021-12-30): remove the raw key from the nodes dict
Fix #192
closed 2022-01-03 by jaykayenn