meshtastic/python #220: Add hwModel to meshtasic cli output

D4rk4, 2022-01-11:
Please add hwModel to meshtastic CLI output In the "My info" section.
It's needed for automatic device type detection for choose with firmware must be flashed via ansible playbook.

mkinney, 2022-01-12:
The data you seek is in the next line of output. In the 'Nodes in mesh' line, it is buried in there under 'user'.

```
% meshtastic --info
Connected to radio

Owner: Unknown 67cc (?CC)
My info: { "myNodeNum": 682584012, "numBands": 13, "firmwareVersion": "1.2.50.41dcfdd", "rebootCount": 53, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 4.6 }
Nodes in mesh:  {'num': 682584012, 'user': {'id': '!28af67cc', 'longName': 'Unknown 67cc', 'shortName': '?CC', 'macaddr': '24:6f:28:af:67:cc', 'hwModel': 'HELTEC_V2_1'}, 'position': {'batteryLevel': 34}, 'lastHeard': 1641951191}  {'num': 682968668, 'user': {'id': '!28b5465c', 'longName': 'Unknown 465c', 'shortName': '?5C', 'macaddr': '24:6f:28:b5:46:5c', 'hwModel': 'HELTEC_V2_1'}, 'position': {'batteryLevel': 48, 'time': 1641855957}, 'lastHeard': 1641855973, 'snr': 12.5}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "positionFlags": 35, "isAlwaysPowered": true }

Channels:
  PRIMARY psk=default { "modemConfig": "Bw125Cr48Sf4096", "psk": "AQ==" }
  SECONDARY psk=secret { "psk": "mcs0IlMwHX/fk8UeToSBQAVI+LDYh9aKMbRgbxHJRcI=", "name": "admin" }

Primary channel URL: https://www.meshtastic.org/d/#CgUYAyIBAQ
Complete URL (includes all channels): https://www.meshtastic.org/d/#CgUYAyIBAQopIiCZyzQiUzAdf9-TxR5OhIFABUj4sNiH1ooxtGBvEclFwioFYWRtaW4
```

See "HELTEC_V2_1"?

Is it easy for you to get at that info? Or would you prefer something easier/different?

D4rk4, 2022-01-12:
@mkinney It's must be also in "My info" section, because I need to get info about the **local node**. Also, I can't trust the output of "Nodes in mesh section".

mkinney, 2022-01-12:
You should be able to trust the Nodes section. The first entry will always be the local node's entry.

The my_info comes from one radio request. The nodes info comes from another radio request.

Here's a snippet that gets you the info you need (at least from a mac):

```
meshtastic --info | grep -o -E "'hwModel': '.*?'" | head -n1
```

This is what it returns for me:

```
'hwModel': 'HELTEC_V2_1'
```

That data could be re-formatted and displayed differently if another command line arg was added or the api were used.

For instance:
```
"""Simple program to demo how to use meshtastic library.
   To run: python examples/get_hw.py
"""

import sys
import meshtastic
import meshtastic.serial_interface

# simple arg check
if len(sys.argv) != 1:
    print(f"usage: {sys.argv[0]}")
    print("Print the hardware model for the local node.")
    sys.exit(3)

iface = meshtastic.serial_interface.SerialInterface()
if iface.nodes:
    for n in iface.nodes.values():
        if n['num'] == iface.myInfo.my_node_num:
            print(n['user']['hwModel'])
iface.close()
```

```
% python3 examples/get_hw.py
HELTEC_V2_1
```

mkinney, 2022-01-12:
That example has just been added to the repo.

D4rk4, 2022-01-12:
@mkinney Thank you for the explanation. But your solution with **grep** not working for me.
This one working fine under Linux, but I can't check how it's working under macOS:
`meshtastic --info | grep -Po "(?<='hwModel':\ ')[^'},]+" | head -1`

mkinney, 2022-01-12:
Mac's grep is kinda old/different... does not have the '-P' option.

```
% meshtastic --info | grep -Po "(?<='hwModel':\ ')[^'},]+" | head -1

grep: invalid option -- P
usage: grep [-abcdDEFGHhIiJLlMmnOopqRSsUVvwXxZz] [-A num] [-B num] [-C[num]]
	[-e pattern] [-f file] [--binary-files=value] [--color=when]
	[--context[=num]] [--directories=action] [--label] [--line-buffered]
	[--null] [pattern] [file ...]
```

closed 2022-02-18 by mkinney