meshtastic/python #234: unable to set wifi_password to a numeric string

kyle2222, 2022-01-15:
firmware: firmware-heltec-v2.1-1.2.52.b63802c.bin
meshtastic --version
1.2.55
meshtastic --set wifi_ap_mode false --set wifi_ssid myWifiAP --set wifi_password 123456789
Connected to radio
Set wifi_ap_mode to false
Set wifi_ssid to myWifiAP
Aborting due to: 123456789 has type int, but expected one of: bytes, unicode

Same thing occurs when I put quotes around the wifi_password string.

mkinney, 2022-01-15:
Can you run `meshtastic --support` and paste that here?

kyle2222, 2022-01-15:
meshtastic --support

If having issues with meshtastic cli or python library
or wish to make feature requests, visit:
https://github.com/meshtastic/Meshtastic-python/issues
When adding an issue, be sure to include the following info:
 System: Darwin
   Platform: macOS-11.6-x86_64-i386-64bit
   Release: 20.6.0
   Machine: x86_64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.2.55
 Executable: /usr/local/bin/meshtastic
 Python: 3.8.2 CPython Clang 12.0.0 (clang-1200.0.31.1)

Please add the output from the command: meshtastic --info

meshtastic --info
Connected to radio

Owner: Unknown 59d8 (?D8)
My info: { "myNodeNum": 3858848216, "numBands": 13, "firmwareVersion": "1.2.52.b63802c", "rebootCount": 1, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true }
Nodes in mesh:  {'num': 3858848216, 'user': {'id': '!e60159d8', 'longName': 'Unknown 59d8', 'shortName': '?D8', 'macaddr': 'e0:e2:e6:01:59:d8', 'hwModel': 'HELTEC_V2_1'}, 'position': {'batteryLevel': 49}, 'lastHeard': 1642283998}  {'num': 2130651664, 'user': {'id': '!7eff2a10', 'longName': 'Unknown 2a10', 'shortName': '?10', 'macaddr': '94:b9:7e:ff:2a:10', 'hwModel': 'HELTEC_V2_1'}, 'position': {}, 'lastHeard': 1642283869, 'snr': 10.0, 'lastReceived': {'from': 2130651664, 'to': 3858848216, 'id': 13671240, 'rxTime': 1642283869, 'rxSnr': 10.0, 'hopLimit': 3, 'fromId': '!7eff2a10', 'toId': '!e60159d8'}, 'hopLimit': 3}  {'num': 2130648944, 'user': {'id': '!7eff1f70', 'longName': 'Unknown 1f70', 'shortName': '?70', 'macaddr': '94:b9:7e:ff:1f:70', 'hwModel': 'HELTEC_V2_1'}, 'position': {}, 'lastHeard': 1642283852, 'snr': 10.0, 'lastReceived': {'from': 2130648944, 'to': 3858848216, 'id': 1640267742, 'rxTime': 1642283852, 'rxSnr': 10.0, 'hopLimit': 3, 'fromId': '!7eff1f70', 'toId': '!e60159d8'}, 'hopLimit': 3}  {'num': 2475426296, 'user': {'id': '!938c01f8', 'longName': 'S20', 'shortName': 'S20', 'macaddr': '44:17:93:8c:01:f8', 'hwModel': 'TBEAM'}, 'lastHeard': 1642283644, 'snr': -19.5}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "wifiSsid": "mywifissid", "wifiPassword": "sekrit", "wifiApMode": true }

Channels:
  PRIMARY psk=default { "modemConfig": "Bw125Cr48Sf4096", "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/d/#CgUYAyIBAQ

mkinney, 2022-01-15:
Thanks. I think there is a bug in `fromStr` function. 

For now, if you put a letter in that password, you should be able to continue.

kyle2222, 2022-01-15:
understood, the problem is that I have many many devices using a numeric password that would need to be reconfigured :-(

mkinney, 2022-01-15:
I'll have a fix shortly. I removed code that I thought was "dead" in the last update.

mkinney, 2022-01-15:
New version for you. Close this issue if it works for you.

Thanks for your help.

kyle2222, 2022-01-15:
confirmed fixed.  You are awesome, thanks so much!

closed 2022-01-15 by kyle2222

geeksville, 2022-01-16:
@mkinney I'm so glad you are here! ;-)