meshtastic/python #261: tunnel failing due to ifconfig

enpass4nt, 2022-02-06:
Hi there!
I am unable to to establish a tunnel on either of my nodes.  Ifconfig is failing and it looks like an issue with the subnet variable in tunnel.py.  Both nodes are assigned "None.x.x" IPs.
Getting the same result on both Ubuntu 18.04.5 LTS Python 3.6.9)  and Raspbian GNU/Linux (Python 3.7.3).

Anyone have any suggestions on how to fix or workaround.  

`$ sudo .local/bin/meshtastic --tunnel
Connected to radio
INFO file:tunnel.py __init__ line:83 Starting IP to mesh tunnel (you must be root for this *pre-alpha* feature to work).  Mesh members:
INFO file:tunnel.py __init__ line:93 Node !25055d9c has IP address None.93.156
INFO file:tunnel.py __init__ line:93 Node !25056dd8 has IP address None.109.216
None.93.156: Unknown host
ifconfig: `--help' gives usage information.
Aborting due to: ifconfig command failed.

 System: Linux
   Platform: Linux-5.4.0-97-generic-i686-with-Ubuntu-18.04-bionic
   Release: 5.4.0-97-generic
   Machine: i686
   Encoding (stdin): UTF-8
   Encoding (stdout): UTF-8
 meshtastic: v1.2.81
 Executable: .local/bin/meshtastic
 Python: 3.6.9 CPython GCC 8.4.0`

mkinney, 2022-02-06:
I have yet to try this.

Did you know about this file? https://github.com/meshtastic/Meshtastic-python/blob/master/extra/meshtastic_tun.rules

enpass4nt, 2022-02-06:
Thank you.  I did not know about that file.  I added my user to netdev and added the rules file, but getting the same result.

I neglected to mention in the original post that I am getting an interface despite the aborted message:
mesh0: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet6 fe80::xxxx:xxxx:xxxx:xxxx  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500

mkinney, 2022-02-06:
Again, I've not tried this yet myself.

I'd start by looking at (or perhaps temp disabling) firewalls. I'd also start looking at the tunnels ("ip a") and perhaps enable `--debug`.

There was quite a good writeup on discourse over this feature, iirc. Might look for that info.

Please share whatever you find.

enpass4nt, 2022-02-12:
After a week of trying countless permutations of hardware, operating systems and python versions, it turns out the solution was embarrassingly simple.   It works when I explicitly set the subnet.  derp.  Thanks again for making this great project!

$ sudo ./meshtastic --subnet 10.115 --tunnel
Connected to radio
INFO file:tunnel.py __init__ line:83 Starting IP to mesh tunnel (you must be root for this *pre-alpha* feature to work).  Mesh members:
INFO file:tunnel.py __init__ line:93 Node !25055d9c has IP address 10.115.93.156
INFO file:tunnel.py __init__ line:93 Node !25056dd8 has IP address 10.115.109.216

enpass4nt, 2022-02-12:
the same setup as detailed here:  https://meshtastic.discourse.group/t/guide-ssh-over-meshtastic/2579

closed 2022-02-13 by mkinney

Mr-Frank-Jaeger, 2022-04-12:
@enpass4nt  i used ur fix and meshtastic seems to be working. However, when i try to ping or use MOSH to set up an ssh connection it just seems to hang or timeout. I used the same guide you posted but keep having the same issue. How did you get over it?

r-tierney, 2024-01-04:
Good find @enpass4nt 

I'm not too familiar with this codebase but at first glance I think it might be this line overriding the default with a None value
https://github.com/meshtastic/python/blob/master/meshtastic/__main__.py#L774C18-L774C18
```python
tunnel.Tunnel(interface, subnet=args.tunnel_net)
```

That None value i believe would get passed here overriding the default: https://github.com/meshtastic/python/blob/master/meshtastic/tunnel.py#L41
```python
def __init__(self, iface, subnet="10.115", netmask="255.255.0.0"):
```

I believe the simple 1 liner fix here would be to first check if that arg is set before using it in the function call
```python
tunnel.Tunnel(interface, subnet=args.tunnel_net) if args.tunnel_net else tunnel.Tunnel(interface)
```

r-tierney, 2024-01-07:
This has been fixed in release version 2.2.17 to use the new tunnel feature simply update to this latest release
```bash
pip3 install --upgrade meshtastic
```