meshtastic/python #265: Settings will not persist after device restart

dfloer, 2022-02-13:
Hardware: Heltec V2.1
Meshtastic: 1.2.55
Python: 3.10.2
OS: Windows 11 Pro build 22000

I'm trying to write configuration settings to the device, for example `meshtastic.exe --port COM3 --setalt x --setlat a.b --setlon c.d`. The settings sometimes take effect, but usually do not. Calling `--export-config` repeatedly will produce _different_ configurations, with settings appearing and disappearing without any configuration changes actually happening. Setting WiFi settings also seems to always fail in 20+ different attempts. This seems to have started when I upgraded to Windows 11.

Someone on Discord mentioned that Windows 11 may reset the serial device, and something definitely is resetting the device. Keeping this in mind, I wondered if it was getting reset _before_ the settings were persisted on the device, so I bumped the sleep in `stream_interface:92` up to 1 second and tried changing the settings again. This time my settings persisted after the device rebooted.

I'm not sure this is the only/best workaround, but it's annoying not being able to set any settings reliably, and this seems to "fix" the problem (or help fix, I tried it 5x and it worked each time). I did not have any issues flashing the device, either using the install script or `meshtastic-flasher`.

If there's anything else I can do to help troubleshoot this, please let me know.

mkinney, 2022-03-04:
@dfloer can you re-check with latest meshtastic `pip install --upgrade meshtastic` and the latest firmware? 1.2.57

I am hoping that the bug fixes in 1.2.57 will fix this issue.

dfloer, 2022-03-04:
If I comment out the increased sleep time for Windows 11, I am still able to reproduce it, but in ~40 or so tests, I only had two fail to completely write the setting to the node properly. So it's better, but was still able to get it to not set everything sent to the device.

closed 2022-07-04 by garthvh