meshtastic/python #273: meshtastic --nodes results in charmap codec error

RicInNewMexico, 2022-02-18:
Command and results:

`Ric@Ric-PC MINGW64 ~/Downloads/firmware
$ meshtastic --nodes
Connected to radio
Aborting due to: 'charmap' codec can't encode characters in position 0-77: character maps to <undefined>

Ric@Ric-PC MINGW64 ~/Downloads/firmware
$ meshtastic --version
1.2.84`

meshtastic --support output:

` System: Windows
   Platform: Windows-7-6.1.7601-SP1
   Release: 7
   Machine: AMD64
   Encoding (stdin): cp1252
   Encoding (stdout): cp1252
 meshtastic: v1.2.84
 Executable: C:\Users\Ric\AppData\Local\Programs\Python\Python37\Scripts\meshtastic
 Python: 3.7.6rc1 CPython MSC v.1916 64 bit (AMD64)`

meshtastic --info output:
`Ric@Ric-PC MINGW64 ~/Downloads/firmware
$ meshtastic --info
Connected to radio

Owner: NMPATP2 (P2)
My info: { "myNodeNum": 2475137016, "hasGps": true, "numBands": 13, "firmwareVersion": "1.2.55.ab96579-d", "rebootCount": 5, "bitrate": 247.13243, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 1.005, "airUtilTx": 0.12677777 }

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300 }

Channels:
  PRIMARY psk=default { "txPower": 20, "REDACTED": "AQ==", "bandwidth": 250, "spreadFactor": 10, "codingRate": 4 }
  SECONDARY psk=secret { "psk": "REDACTED", "name": "admin" }

Primary channel URL: https://www.meshtastic.org/d/#REDACTED
Complete URL (includes all channels): https://www.meshtastic.org/d/#REDACTED`

mkinney, 2022-02-18:
This issue might help: https://github.com/meshtastic/Meshtastic-gui-installer/pull/33

RicInNewMexico, 2022-02-18:
Functions correctly on ubuntu:

meshtastic --support info:
```
 System: Linux
   Platform: Linux-4.4.190-1233-rockchip-ayufan-gd3f1be0ed310-aarch64-with-Ubuntu-18.04-bionic
   Release: 4.4.190-1233-rockchip-ayufan-gd3f1be0ed310
   Machine: aarch64
   Encoding (stdin): UTF-8
   Encoding (stdout): UTF-8
 meshtastic: v1.2.84
 Executable: /home/rock64/.local/bin/meshtastic
 Python: 3.6.9 CPython GCC 8.4.0
```
I suspect it's not handling the encoding differences on some platforms well.

RicInNewMexico, 2022-02-18:
This issue, for me anyway, seems to be confined to Mingw64's bash terminal. I discovered that --nodes works as expected within the windows command prompt.

mkinney, 2022-03-04:
I'm going to close this issue since it is recommended to use a command prompt.

closed 2022-03-04 by mkinney