meshtastic/python #279: Releases are not archives

davidmotson, 2022-02-19:
Since version 1.2.73 the release artifacts for the standalone meshtastic-python have been executables, and missing their file extensions, instead of archives as documented.

sachaw, 2022-02-19:
@mkinney Another think to think about: maybe remove the readme.txt and either:
1. Have a note on reach release (not so desirable) 
2. Add a note/admontion on the docs site

mkinney, 2022-02-26:
Wow. Those instructions were out of date. I just did a PR: https://github.com/meshtastic/Meshtastic/pull/263

The https://github.com/meshtastic/Meshtastic-python/blob/master/standalone_readme.txt is in the repo and added on each release. I think there needs to be something there to point to the website, which this file is a pretty minimal version of that.

closed 2022-02-26 by mkinney