meshtastic/python #285: Meshtastic-python pub-sub-example not connecting, where mosquito_pub and _sub succeed

bradshawlupton, 2022-02-28:
::::::::::::
::::::::::::
TEST mosquitto sub pub to localhost and test.mosquitto.org (WORKS)
::::::::::::
::::::::::::

mosquitto_pub -h localhost                 -t FRED -m pub.to.localhost
pub.to.test.localhost
mosquitto_pub -h test.mosquitto.org -t FRED -m pub.to.test.mosquitto.org
pub.to.test.mosquitto.org

::::::::::::
::::::::::::
TEST Meshtastic_python examples (NOT WORKING, YET)
::::::::::::
::::::::::::

timeout 30 python pub_sub_example.py.    localhost
     Error: Could not connect to localhost

timeout 30 python pub_sub_example.py.    test.mosquitto.org
     timed out
timeout 30 python pub_sub_example2.py localhost
     Error: Could not connect to localhost

timeout 30 python pub_sub_example2.py test.mosquitto.org
     timed out

--------------
-------------- test script to test mosquito, then meshtastic pub_sub examples
--------------

#!/bin/bash

cd /home/pi/SW/Meshtastic-python/examples

echo ::::::::::::
echo ::::::::::::
echo TEST mosquitto sub pub to localhost and test.mosquitto.org \(WORKS\)
echo ::::::::::::
echo ::::::::::::
echo

mosquitto_sub -h localhost -t FRED &
sleep 1
echo mosquitto_pub -h localhost -t FRED -m pub.to.localhost
mosquitto_pub -h localhost -t FRED -m pub.to.test.localhost
sleep 1

killall mosquitto_sub

mosquitto_sub -h test.mosquitto.org -t FRED &
sleep 1
echo mosquitto_pub -h test.mosquitto.org -t FRED -m pub.to.test.mosquitto.org
mosquitto_pub -h test.mosquitto.org -t FRED -m pub.to.test.mosquitto.org
sleep 1

killall mosquitto_sub

echo
echo ::::::::::::
echo ::::::::::::
echo TEST Meshtastic_python examples  \(NOT WORKING, YET\)
echo ::::::::::::
echo ::::::::::::
echo 
echo timeout 30 python pub_sub_example.py localhost
timeout 30 python pub_sub_example.py localhost
echo
sleep 1

echo timeout 30 python pub_sub_example.py test.mosquitto.org
timeout 30 python pub_sub_example.py test.mosquitto.org
echo
sleep 1

echo timeout 30 python pub_sub_example2.py localhost
timeout 30 python pub_sub_example.py localhost
echo
sleep 1

echo timeout 30 python pub_sub_example2.py test.mosquitto.org
timeout 30 python pub_sub_example.py test.mosquitto.org
echo
sleep 1

mkinney, 2022-02-28:
I would probably not be using the 1.3.0.c97541d4-d branch quite yet. It is very unstable and very active.

I'd switch to the 1.2.56.596a73c until 1.3 is more stable.

mkinney, 2022-02-28:
Be sure to checkout https://meshtastic.org/docs/software/other/mqtt#mini-tutorial-on-how-to-get-up-and-running-with-mosquitto-on-a-mac

garthvh, 2023-03-19:
Stale, closing

closed 2023-03-19 by garthvh