meshtastic/python #301: Device error when setting wifi password too short

mc-hamster, 2022-03-15:
Any chance for string validation that it's at least 8 characters in length?

Thanks mate!

---

raldi — Today at 7:39 PM
Looks like the problem was that the wifi password was too short
When that happens, the UI thinks it's set the SSID and password, but it actually hasn't; the driver uses its default, which in my case was ESP_xxxx
What do you think about changing https://github.com/meshtastic/Meshtastic-device/blob/master/src/DebugConfiguration.h#L49 to #error DEBUG_MSG should always be defined to something?
MC Hamster — Today at 7:40 PM
Oh! How did you set the password?
raldi — Today at 7:40 PM
Originally I was trying to set it to M
but that's too short
So I just turned off the password altogether
MC Hamster — Today at 7:41 PM
Did you set the password over the python cli or some other method?
raldi — Today at 7:41 PM
Python CLI

mkinney, 2022-03-17:
Add to 1.2 or 1.3 or both? @mc-hamster

mkinney, 2022-03-21:
Added to both branches. Released new versions:

https://github.com/meshtastic/Meshtastic-python/releases/tag/1.2.93
https://github.com/meshtastic/Meshtastic-python/releases/tag/1.3alpha.8

closed 2022-03-21 by mkinney