meshtastic/python #317: [Bug]: Issues setting psk key via serial

rwcts, 2022-03-31:


mkinney, 2022-03-31:
The actual string must be in the "0x<HEX><HEX>..." and it must be a specific length, depending on which level you want. (16 bytes for AES128 or 32 for AES256)

mkinney, 2022-03-31:
Note: The string displayed when using `--info` is not the same value that would be used here.

rwcts, 2022-03-31:
> 

I have confirmed that the key used is not the one showing in --info so that is correct. I did --ch-set psk random --ch-index 0 and then logged into the client.meshtastic.org and looked at the key it generated. The key was +ZDoJyvLSPhCBuT6IKCMiOE1l6pJc+VN19S4AGhyFi0=  

So when using the python serial interface (command prompt) and then entering in the below command, it should not work as its not HEX. If that is the case then its a "me" problem not a bug.

meshtastic --ch-set psk +ZDoJyvLSPhCBuT6IKCMiOE1l6pJc+VN19S4AGhyFi0= --info --ch-index 0 --port COM9

mkinney, 2022-03-31:
The way the `--ch-set psk` is written, is it expects the values to be in a different format than the string representations that you have listed.

It's not really a "you" issue. It's just the way the code is currently written. To allow a string format, it would require some coding.

closed 2022-03-31 by garthvh

sachaw, 2022-09-29:
The web UI will allow you to copy/paste keys between devices