meshtastic/python #318: Unable to talk to RAK5005-O over USB at all

joshbowyer, 2022-04-04:
I got the RAK5005-O working, but when I plug it into my computer via USB and try to run any command with the meshtastic python program, it always returns with a connection timed out error. I installed using pip. RAK Meshtastic firmware is the latest.

garthvh, 2022-04-04:
On windows? Do you have a battery connected?

joshbowyer, 2022-04-04:
Sorry, on Debian and there is a battery connected yes

garthvh, 2022-04-04:
Latest being 1.2.59? Are you using a Raspberry Pi?

joshbowyer, 2022-04-04:
Ill have to double check when I get home but I ran the meshtastic-updater yesterday and it said firmware is the latest. And this is not from a RPi, just my laptop

mkinney, 2022-04-04:
Can you run `meshtastic --info --debug` and share the results? Also, did you add your user to the dialout group and re-login?

joshbowyer, 2022-04-04:
I did add my user to dialout and rebooted actually; that error no longer appears. I will try that debug command in a couple hours and will report my findings.

joshbowyer, 2022-04-04:
https://pastebin.com/JQdMMDmY

It looks like a looping error

joshbowyer, 2022-04-04:
The weird thing is it looks like just this one time it did print the info, but in the subsequent times I ran the same command (with debug) it failed to do so and just hung there

joshbowyer, 2022-04-04:
I wonder if this is the problem, I have not installed any driver as mentioned here https://meshtastic.org/docs/getting-started/flashing-firmware

However, my board has a RAK4630 which the documentation says is NRF, but the Linux NRF driver link is dead... do I need to install it, and if so, where can I get it?

joshbowyer, 2022-04-04:
Ok, I found the corresponding git repo for that page: https://github.com/WCHSoftGroup/ch343ser_linux (WCH = wch-ih.com)

I loaded the driver and now I am able to communicate with the board using meshtastic-python.

closed 2022-04-04 by joshbowyer

garthvh, 2022-04-05:
This driver has been in the kernel on most linux versions for a while, what distro are you running?

joshbowyer, 2022-04-05:
PureOS, so maybe they stripped it out? They dont do nonfree stuff apparently...