meshtastic/python #319: Heltec Lora 32 V2.1 when programming it erases other data

rwcts, 2022-04-09:
This is the only device connected to the computer via USB Cable.

Using Python
--support output
If having issues with meshtastic cli or python library
or wish to make feature requests, visit:
https://github.com/meshtastic/Meshtastic-python/issues
When adding an issue, be sure to include the following info:
 System: Windows
   Platform: Windows-10-10.0.22000-SP0
   Release: 10
   Machine: AMD64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.3a11
 Executable: C:\Users\User\appdata\local\programs\python\python310\scripts\meshtastic
 Python: 3.10.3 CPython MSC v.1929 64 bit (AMD64)

Below I have the output of the --info (some items hidden) and then commands made and then --info after showing the issue

--------------------------output-------------------------
C:\Users\User\Downloads>meshtastic --info --port COM9
Connected to radio

Owner: ACTUALNAMEHIDDEN (EC)
My info: { "myNodeNum": 2130654944, "firmwareVersion": "1.3.5.e5b19fd", "rebootCount": 51, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 5.6383333, "airUtilTx": 0.09397222 }
Nodes in mesh:  {'num': 2130654944, 'user': {'id': '!7eff36e0', 'longName': 'ACTUALNAMEHIDDEN', 'shortName': 'EC', 'macaddr': '94:b9:7e:ff:36:e0', 'hwModel': 'HELTEC_V2_1'}, 'position': {}, 'telemetry': {}}  {'num': 4145956180, 'user': {'id': '!f71e4554', 'longName': 'ACTUALNAMEHIDDEN', 'shortName': 'E8', 'macaddr': '30:c6:f7:1e:45:54', 'hwModel': 'TBEAM'}, 'snr': 12.5, 'lastHeard': 1649533986}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "positionFlags": 35 }

Channels:
  PRIMARY psk=secret { "modemConfig": "LongSlow", "psk": "ACTUAL KEY HIDDEN", "uplinkEnabled": true }
  SECONDARY psk=secret { "psk": "ACTUAL KEY HIDDEN", "name": "admin" }

Primary channel URL: ACTUAL LINK HIDDEN
Complete URL (includes all channels): ACTUAL LINK HIDDEN

C:\Users\User\Downloads>meshtastic --ch-set uplink_enabled true --ch-index 1 --port COM9
Connected to radio
Set uplink_enabled to true
Writing modified channels to device

C:\Users\User\Downloads>meshtastic --ch-set downlink_enabled true --ch-index 1 --port COM9
Connected to radio
Set downlink_enabled to true
Writing modified channels to device

C:\Users\User\Downloads>meshtastic --info --port COM9
Connected to radio

Owner: ACTUALNAMEHIDDEN (EC)
My info: { "myNodeNum": 2130654944, "firmwareVersion": "1.3.5.e5b19fd", "rebootCount": 55, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 1.3399999, "airUtilTx": 0.022333333 }
Nodes in mesh:  {'num': 2130654944, 'user': {'id': '!7eff36e0', 'longName': 'ACUTALNAMEHIDDEN', 'shortName': 'EC', 'macaddr': '94:b9:7e:ff:36:e0', 'hwModel': 'HELTEC_V2_1'}, 'position': {}, 'telemetry': {}}  {'num': 4145956180, 'user': {'id': '!f71e4554', 'longName': 'ACTUALNAMEHIDDEN', 'shortName': 'E8', 'macaddr': '30:c6:f7:1e:45:54', 'hwModel': 'TBEAM'}, 'snr': 12.5, 'lastHeard': 1649533986}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "positionFlags": 35 }

Channels:
  PRIMARY psk=default { "modemConfig": "LongFast", "psk": "AQ==" }

Primary channel URL: ACTUAL LINK HIDDEN

rwcts, 2022-04-09:
Previous to starting to program the unit I did a fresh firmware flash

mkinney, 2022-04-09:
That is a high reboot count.

rwcts, 2022-04-09:
> That is a high reboot count.

Yes during programming my experience has been that each command it will reboot. This has been a consistent behavior I have seen with all 10 of the heltec's during setup.

kevinb0, 2022-04-14:
I'm seeing the same thing, here's my info. 

<html><head>
  
  <meta http-equiv='Content-Type' content='text/html; charset=UTF-8'/>
  <title>mintty screen dump</title>
  <link rel='stylesheet' type='text/css' href='xterm.css'/>
  <link rel='stylesheet' type='text/css' href='mintty.css'/>
  <style type='text/css'>
  body, pre { margin: 0; padding: 0; }
  pre { font-family: inherit; }
  #vt100 {
    float: left;
    border: 0px solid;
    padding: 1px;
    line-height: 116%;
    font-size: 9pt;
    font-family: 'Lucida Console', 'Lucida Console ', 'Consolas';
    color: #BFBFBF;
    background-color: #000000;
    }
  .bd { font-weight: bold }
  .it { font-style: italic }
  .ul { text-decoration-line: underline }
  .st { text-decoration-line: line-through }
  .lu { text-decoration-line: line-through underline }
  .fg-color0 { color: #000000 } .bg-color0 { background-color: #000000 }
  .fg-color1 { color: #BF0000 } .bg-color1 { background-color: #BF0000 }
  .fg-color2 { color: #00BF00 } .bg-color2 { background-color: #00BF00 }
  .fg-color3 { color: #BFBF00 } .bg-color3 { background-color: #BFBF00 }
  .fg-color4 { color: #0000BF } .bg-color4 { background-color: #0000BF }
  .fg-color5 { color: #BF00BF } .bg-color5 { background-color: #BF00BF }
  .fg-color6 { color: #00BFBF } .bg-color6 { background-color: #00BFBF }
  .fg-color7 { color: #BFBFBF } .bg-color7 { background-color: #BFBFBF }
  .fg-color8 { color: #404040 } .bg-color8 { background-color: #404040 }
  .fg-color9 { color: #FF4040 } .bg-color9 { background-color: #FF4040 }
  .fg-color10 { color: #40FF40 } .bg-color10 { background-color: #40FF40 }
  .fg-color11 { color: #FFFF40 } .bg-color11 { background-color: #FFFF40 }
  .fg-color12 { color: #6060FF } .bg-color12 { background-color: #6060FF }
  .fg-color13 { color: #FF40FF } .bg-color13 { background-color: #FF40FF }
  .fg-color14 { color: #40FFFF } .bg-color14 { background-color: #40FFFF }
  .fg-color15 { color: #FFFFFF } .bg-color15 { background-color: #FFFFFF }
  .cursor { background-color: #BFBFBF }
  .font10 { font-family: 'F25 Blackletter Typewriter' }
  </style>
  <script>
  var b1 = 500; var b2 = 300;
  function visib (tag, state, timeout) {
    var bl = document.getElementsByName(tag);
    var vv; if (state) vv = 'visible'; else vv = 'hidden';
    var i;
    for (i = 0; i < bl.length; i++) {
      bl[i].style.visibility = vv;
    }
    window.setTimeout ('visib ("' + tag + '", ' + !state + ', ' + timeout + ')', timeout);
  }
  function setup () {
    window.setTimeout ('visib ("blink", 0, b1)', b1);
    window.setTimeout ('visib ("rapid", 0, b2)', b2);
  }
  </script>
</head>

<body onload='setup();'>
  

https://github.com/meshtastic/Meshtastic-python/issues                                                                                                                                                                                                                          When adding an issue, be sure to include the following info:                                                                                                                                                                                                                     System: Windows                                                                                                                                                                                                                                                                   Platform: Windows-10-10.0.19044-SP0                                                                                                                                                                                                                                             Release: 10                                                                                                                                                                                                                                                                     Machine: AMD64                                                                                                                                                                                                                                                                  Encoding (stdin): cp1252                                                                                                                                                                                                                                                        Encoding (stdout): cp1252                                                                                                                                                                                                                                                     meshtastic: v1.2.93                                                                                                                                                                                                                                                             Executable: C:\Users\Kevin\AppData\Local\Programs\Python\Python310\Scripts\meshtastic                                                                                                                                                                                           Python: 3.10.2 CPython MSC v.1929 64 bit (AMD64)
--

</body>
</html>

The thing that I note, is that when you load a custom file, it automatically gets reset. for example if you do the following
meshtastic --configure settings.yaml (where it's just a generic file) without anything, when restarted, the settigns will be set back to default. However, if I do them in the same line (meshtastic --configure --info), you can see that they are correctly set. Check out this output on the file 

[image]

--

rwcts, 2022-04-29:
I just tested the newest 1.3.10 version and had the same issue and this falls along the similar lines of another ticket recently opened. https://github.com/meshtastic/Meshtastic-device/issues/1396

C:\Users\User\Downloads>meshtastic --ch-set name CCC --ch-index 0 --port COM9
Connected to radio
Set name to CGH
Writing modified channels to device

C:\Users\User\Downloads>meshtastic --ch-set uplink_enabled true --ch-index 0 --port COM9
Connected to radio
Set uplink_enabled to true
Writing modified channels to device

C:\Users\User\Downloads>meshtastic --ch-set downlink_enabled true --ch-index 0 --port COM9
Connected to radio
Set downlink_enabled to true
Writing modified channels to device

AFTER THE ABOVE COMMAND is when it reset the region however in the past on version 1.3.7? It did it when you set the PSK key or the URL

C:\Users\User\Downloads>meshtastic --ch-set psk random --ch-index 0 --port COM9
Connected to radio
Writing modified channels to device

C:\Users\User\Downloads>meshtastic --info --port COM9
Connected to radio

Owner: RWS2 (E2)
My info: { "myNodeNum": 2130654728, "firmwareVersion": "1.3.10.cc2a84a", "rebootCount": 15, "bitrate": 4.2721176, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "maxChannels": 8, "hasWifi": true }
Nodes in mesh:  {'num': 2130654728, 'user': {'id': '!7eff3608', 'longName': 'RWS2', 'shortName': 'E2', 'macaddr': ‘94:b9:7e:fl:16:99’, 'hwModel': 'HELTEC_V2_1'}, 'position': {}, 'deviceMetrics': {'channelUtilization': 15.781667, 'airUtilTx': 0.2630278}}

Preferences: {}

Channels:

Primary channel URL: https://www.meshtastic.org/d/#

C:\Users\User\Downloads>meshtastic --ch-set modem_config LongSlow --ch-index 0 --port COM9
Connected to radio
Set modem_config to LongSlow
Writing modified channels to device

C:\Users\User\Downloads>meshtastic --info --port COM9
Connected to radio

Owner: RWS2 (E2)
My info: { "myNodeNum": 2130654728, "firmwareVersion": "1.3.10.cc2a84a", "rebootCount": 17, "bitrate": 4.2721176, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "maxChannels": 8, "hasWifi": true }
Nodes in mesh:  {'num': 2130654728, 'user': {'id': '!7eff3608', 'longName': 'RWS2', 'shortName': 'E2', 'macaddr': ‘94:b9:7e:fl:16:99’, 'hwModel': 'HELTEC_V2_1'}, 'position': {}, 'deviceMetrics': {'channelUtilization': 33.313335, 'airUtilTx': 0.81824994}}

Preferences: {}

Channels:

Primary channel URL: https://www.meshtastic.org/d/#

C:\Users\User\Downloads>meshtastic --support

If having issues with meshtastic cli or python library
or wish to make feature requests, visit:
https://github.com/meshtastic/Meshtastic-python/issues
When adding an issue, be sure to include the following info:
 System: Windows
   Platform: Windows-10-10.0.22000-SP0
   Release: 10
   Machine: AMD64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.3a16
 Executable: C:\Users\User\appdata\local\programs\python\python310\scripts\meshtastic
 Python: 3.10.4 CPython MSC v.1929 64 bit (AMD64)

Please add the output from the command: meshtastic --info

rwcts, 2022-04-29:
This is the output from the initial flash on todays testing if its helpful

C:\Users\User\Downloads>cd firmware-1.3.10.cc2a84a

C:\Users\User\Downloads\firmware-1.3.10.cc2a84a>device-install.bat -f firmware-heltec-v2.1-1.3.10.cc2a84a.bin
Trying to flash update firmware-heltec-v2.1-1.3.10.cc2a84a.bin, but first erasing and writing system information"
esptool.py v3.3
Found 5 serial ports
Serial port COM9
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:b9:7e:fl:16:99
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 7.6s
Hard resetting via RTS pin...
esptool.py v3.3
Found 5 serial ports
Serial port COM9
Connecting......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:b9:7e:fl:16:99
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x0000ffff...
Compressed 61440 bytes to 11950...
Wrote 61440 bytes (11950 compressed) at 0x00001000 in 0.5 seconds (effective 901.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
esptool.py v3.3
Found 5 serial ports
Serial port COM9
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting......
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:b9:7e:fl:16:99
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00390000 to 0x003fffff...
Compressed 458752 bytes to 272497...
Wrote 458752 bytes (272497 compressed) at 0x00390000 in 3.8 seconds (effective 962.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
esptool.py v3.3
Found 5 serial ports
Serial port COM9
Connecting.......
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:b9:7e:fl:16:99
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00010000 to 0x001abfff...
Compressed 1686112 bytes to 988338...
Wrote 1686112 bytes (988338 compressed) at 0x00010000 in 13.9 seconds (effective 973.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
C:\Users\User\Downloads\firmware-1.3.10.cc2a84a>

rwcts, 2022-04-29:
After the above firmware flash here is an example of output including commands and the result after the last command is the speed going from LongFast to VLongFast. This is a little off topic but shows a sign of change that should not be occuring?

C:\Users\User\Downloads\firmware-1.3.10.cc2a84a>meshtastic --set region US --port COM9
Connected to radio
Set region to US
Writing modified preferences to device

C:\Users\User\Downloads\firmware-1.3.10.cc2a84a>meshtastic --info --port COM9
Connected to radio

Owner: Unknown 3608 (?08)
My info: { "myNodeNum": 2130654728, "firmwareVersion": "1.3.10.cc2a84a", "rebootCount": 3, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "maxChannels": 8, "hasWifi": true, "channelUtilization": 0.97166663, "airUtilTx": 0.016194444 }
Nodes in mesh:  {'num': 2130654728, 'user': {'id': '!7eff3608', 'longName': 'Unknown 3608', 'shortName': '?08', 'macaddr': '94:b9:7e:fl:16:99', 'hwModel': 'HELTEC_V2_1'}, 'position': {}, 'deviceMetrics': {}}

Preferences: {}

Channels:

Primary channel URL: https://www.meshtastic.org/d/#

C:\Users\User\Downloads\firmware-1.3.10.cc2a84a>meshtastic --set-owner RWS2 --port COM9
Connected to radio
Setting device owner to RWS2

C:\Users\User\Downloads\firmware-1.3.10.cc2a84a>meshtastic --info --port COM9
Connected to radio

Owner: RWS2 (RW-)
My info: { "myNodeNum": 2130654728, "firmwareVersion": "1.3.10.cc2a84a", "rebootCount": 5, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "maxChannels": 8, "hasWifi": true, "channelUtilization": 1.9850001, "airUtilTx": 0.033083335 }
Nodes in mesh:  {'num': 2130654728, 'user': {'id': '!7eff3608', 'longName': 'RWS2', 'shortName': 'RW-', 'macaddr': '94:b9:7e:fl:16:99', 'hwModel': 'HELTEC_V2_1'}, 'position': {}, 'deviceMetrics': {'channelUtilization': 2.08, 'airUtilTx': 0.034666665}}

Preferences: {}

Channels:

Primary channel URL: https://www.meshtastic.org/d/#

C:\Users\User\Downloads\firmware-1.3.10.cc2a84a>meshtastic --set-owner-short E2 --port COM9
Connected to radio
Setting device owner short to E2

C:\Users\User\Downloads\firmware-1.3.10.cc2a84a>meshtastic --info --port COM9
Connected to radio

Owner: RWS2 (E2)
My info: { "myNodeNum": 2130654728, "firmwareVersion": "1.3.10.cc2a84a", "rebootCount": 7, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20300, "maxChannels": 8, "hasWifi": true, "channelUtilization": 0.97166663, "airUtilTx": 0.016194444 }
Nodes in mesh:  {'num': 2130654728, 'user': {'id': '!7eff3608', 'longName': 'RWS2', 'shortName': 'E2', 'macaddr': '94:b9:7e:fl:16:99', 'hwModel': 'HELTEC_V2_1'}, 'position': {}, 'deviceMetrics': {'channelUtilization': 2.08, 'airUtilTx': 0.034666665}}

Preferences: {}

Channels:

Primary channel URL: https://www.meshtastic.org/d/#

C:\Users\User\Downloads\firmware-1.3.10.cc2a84a>meshtastic --ch-set id 1111 --ch-index 0 --port COM9
Connected to radio
Set id to 1111
Writing modified channels to device

C:\Users\User\Downloads\firmware-1.3.10.cc2a84a>

rwcts, 2022-04-29:
I just attempted several times and can replicate this issue everytime without issue. When running "meshtastic --ch-set downlink_enabled true --ch-index 0 --port COM9" it will change the speed from LongSlow to VLongSlow and when running "meshtastic --ch-set psk random --ch-index 0 --port COM" AND "meshtastic --ch-set psk 0x2A47...REMAININGisHIDDEN --info --ch-index 0 --port COM9" it will unset the region

mkinney, 2022-04-29:
I confirm that 1.3 is broken in this regard.

rwcts, 2022-04-29:
Any other info I can send that would be helpful? If so just reach out. Thanks!

> I confirm that 1.3 is broken in this regard.

mkinney, 2022-04-29:
Not at this time, unless you can figure it out. ;-)

I suspect the issue is with protobufs or something recently added in the 1.3 stuffs.

garthvh, 2022-06-20:
This is fixed in 1.3.20

Both --nodes and --info are working, there is data missing.

you can also run

--shutdown
--reset

And save region.

closed 2022-06-20 by garthvh