meshtastic/python #321: Mestastic 1.3a11 don't set or get canned messages

amerinoj, 2022-04-10:
Meshtastic-python 1.3.a11 don't use a good value to get or set the canned messagess

Test with last firmware version 1.3.5.59
./run.sh --info
Owner: Unknown 620c (?0C)
My info: { "myNodeNum": 633561612, "firmwareVersion": "1.3.5.595e5f4f", "rebootCount": 273, "bitrate": 74.341286, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true }
Nodes in mesh: {'num': 633561612, 'user': {'id': '!25c3620c', 'longName': 'Unknown 620c', 'shortName': '?0C', 'macaddr': '4c:75:25:c3:62:0c', 'hwModel': 'TLORA_V2_1_1p6'}, 'position': {}}

./meshtastic --version
1.3a11

./meshtastic-flasher

Writing preferences to device
Set canned_message_module_enabled to True
Set canned_message_module_allow_input_source to _any
Set canned_message_module_send_bell to True
UserPreferences does not have an attribute called canned_message_module_messages, so you can not set it.

./meshtastic --set-canned-message "hi|bye"
Connected to radio
Setting canned plugin message to hi|bye
Aborting due to: 'AdminMessage' object has no attribute 'set_canned_message_plugin_part1'
./meshtastic --get-canned-message
Connected to radio

Aborting due to: 'AdminMessage' object has no attribute 'get_canned_message_plugin_part1_request'

#############################################################################################
The old version 1.2.93 work fine
./run.sh --version
rm: no se puede borrar 'log_*': No existe el archivo o el directorio
1.2.93
./run.sh --set-canned-message "hi|bye"
Connected to radio
Setting canned plugin message to hi|bye

############################################################################################
Reviewing the firmware code

Python 1.3.a11 get_canned_message_plugin_part1_request
Firmware version 1.3.5. get_canned_message_module_part1_request

(variant,**get_canned_message_module_part1_request,get_canned_message_module_part1_request**), 36) \ X(a, STATIC, ONEOF, STRING, (variant,get_canned_message_module_part1_response,get_canned_message_module_part1_response), 37) \ X(a, STATIC, ONEOF, BOOL, (variant,get_canned_message_module_part2_request,get_canned_message_module_part2_request), 38) \ X(a, STATIC, ONEOF, STRING, (variant,get_canned_message_module_part2_response,get_canned_message_module_part2_response), 39) \ X(a, STATIC, ONEOF, BOOL, (variant,get_canned_message_module_part3_request,get_canned_message_module_part3_request), 40) \ X(a, STATIC, ONEOF, STRING, (variant,get_canned_message_module_part3_response,get_canned_message_module_part3_response), 41) \ X(a, STATIC, ONEOF, BOOL, (variant,get_canned_message_module_part4_request,get_canned_message_module_part4_request), 42) \ X(a, STATIC, ONEOF, STRING, (variant,get_canned_message_module_part4_response,get_canned_message_module_part4_response), 43) \ X(a, STATIC, ONEOF, STRING, (variant,set_canned_message_module_part1,set_canned_message_module_part1), 44) \ X(a, STATIC, ONEOF, STRING, (variant,set_canned_message_module_part2,set_canned_message_module_part2), 45) \ X(a, STATIC, ONEOF, STRING, (variant,set_canned_message_module_part3,set_canned_message_module_part3), 46) \ X(a, STATIC, ONEOF, STRING, (variant,set_canned_message_module_part4,set_canned_message_module_part4), 47) \

amerinoj, 2022-04-10:
Ok, i found the problem

node.py is doing reference to "get_canned_message_plugin_" and in the admin_pb2.py and firmware is doing reference to "get_canned_message_module_"

replacement the entries "get canned_message plugin" to get_canned_message_module_" in node.py it's work

mkinney, 2022-04-11:
Just released a new alpha version. Can you test to see if your issue is resolved?

amerinoj, 2022-04-12:
Yes, now is working fine.

closed 2022-04-12 by amerinoj