meshtastic/python #331: When using standalone CLI, --nodes gives: No module named 'timeago.locales.en'

jasonseritt, 2022-05-05:
Not sure if user error or actual issue, here's what I'm seeing, node lat/longs redacted:

**>meshtastic --info**
Connected to radio

Owner: T-Beam 1 (TB1)
My info: { "myNodeNum": 621084160, "hasGps": true, "numBands": 13, "firmwareVersion": "1.2.63.9879494", "rebootCount": 20, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 29.106668, "airUtilTx": 2.8896668 }
Nodes in mesh:  {'num': 621084160, 'user': {'id': '!2504fe00', 'longName': 'T-Beam 1', 'shortName': 'TB1', 'macaddr': 'xxxx', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': xxxx, 'longitudeI':xxxx, 'altitude': 146, 'batteryLevel': 100, 'time': 1651725985, 'locationSource': 'LOCSRC_GPS_INTERNAL', 'posTimestamp': 1651725964, 'altitudeHae': 117, 'altGeoidSep': -28, 'PDOP': 197, 'fixType': 3, 'satsInView': 7, 'latitude': xxxx, 'longitude': xxxx}, 'lastHeard': 1651725985, 'snr': 10.5}  {'num': 621168712, 'user': {'id': '!25064848', 'longName': 'Router', 'shortName': 'RTR', 'macaddr': '58:bf:25:06:48:48', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': xxxx, 'longitudeI': xxxx, 'altitude': 156, 'time': 1651725395, 'latitude': xxxx, 'longitude': xxxx}, 'lastHeard': 1651725371, 'snr': 9.0, 'lastReceived': {'from': 621168712, 'to': 4294967295, 'id': 100849197, 'rxTime': 1651725371, 'rxSnr': 9.0, 'hopLimit': 3, 'fromId': '!25064848', 'toId': '^all'}, 'hopLimit': 3}  {'num': 621093584, 'user': {'id': '!250522d0', 'longName': 'T-Beam 2', 'shortName': 'TB2', 'macaddr': '58:bf:25:05:22:d0', 'hwModel': 'TBEAM'}, 'position': {'latitudeI': xxxx, 'longitudeI': xxxx, 'time': 1651725839, 'latitude':xxxx, 'longitude': xxxx}, 'lastHeard': 1651725956, 'snr': 11.5, 'lastReceived': {'from': 621093584, 'to': 4294967295, 'id': 1000591681, 'rxTime': 1651725956, 'rxSnr': 11.5, 'hopLimit': 3, 'fromId': '!250522d0', 'toId': '^all'}, 'hopLimit': 3}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "wifiPassword": "sekrit", "region": "US", "chargeCurrent": "MA550", "positionBroadcastSmart": true, "locationShare": "LocEnabled", "gpsOperation": "GpsOpMobile", "positionFlags": 35 }

Channels:
  PRIMARY psk=default { "modemConfig": "Bw125Cr48Sf4096", "psk": "AQ==" }
  SECONDARY psk=secret { "psk": "v2sFhsAyJiBy3dyixz7P+p2NBqX+Ingxf7QZoH/hAWc=", "name": "admin" }

Primary channel URL: https://www.meshtastic.org/d/#CgUYAyIBAQ
Complete URL (includes all channels): https://www.meshtastic.org/d/#CgUYAyIBAQopIiC_awWGwDImIHLd3KLHPs_6nY0Gpf4ieDF_tBmgf-EBZyoFYWRtaW4

**>meshtastic --nodes**
Connected to radio
Aborting due to: No module named 'timeago.locales.en'

**>pip install timeago**
Requirement already satisfied: timeago in c:\users\administrator\anaconda3\lib\site-packages (1.0.15)

**>meshtastic --support**

If having issues with meshtastic cli or python library
or wish to make feature requests, visit:
https://github.com/meshtastic/Meshtastic-python/issues
When adding an issue, be sure to include the following info:
 System: Windows
   Platform: Windows-10-10.0.19043-SP0
   Release: 10
   Machine: AMD64
   Encoding (stdin): utf-8
   Encoding (stdout): utf-8
 meshtastic: v1.2.93
 Executable: meshtastic
 Python: 3.9.10 CPython MSC v.1929 64 bit (AMD64)

DaveTheRave2, 2022-05-28:
I am getting the same error. I looked at the documentation for timeago and it looks like the call should be timeago.locale.en rather than timeago.locales.en. Unfortunately, I am not proficient enough in coding to make the change to the software. Hopefully this will be assigned to a real programmer to make the fix as this would be a really useful feature to have working.

garthvh, 2022-11-14:
Can't replicate

closed 2022-11-14 by garthvh

basilyohannan, 2023-01-04:
Connected to radio
Aborting due to: No module named 'timeago.locales.en'

menguinponkey, 2023-02-10:
Please reopen this issue, I'm having the same problem.

jp-bennett, 2023-07-26:
@garthvh I'm currently seeing this too, when running the downloaded build from the GH release. The version installed via pip works fine. Anything I can do to help pinpoint what's causing this?