meshtastic/python #333: meshtastic CLI hangs on MAC

shivharis, 2022-05-05:
The following code on MAC Catalina 10.15.7 hangs in open. Not sure why an open() would hang - it should either open the device or return with error. Hang is not expected!

        if platform.system() != 'Windows':
            with open(self.devPath, encoding='utf8') as f:
                attrs = termios.tcgetattr(f)
                attrs[2] = attrs[2] & ~termios.HUPCL
                termios.tcsetattr(f, termios.TCSAFLUSH, attrs)
                f.close()
            time.sleep(0.1)

(BTW: platform.system() return 'Darwin' for Catalina 10.15.7) Discovered hang using debug log print statements.

closed 2022-11-14 by garthvh

holdenweb, 2024-04-08:
Without a reference to the commit that fixed this issue, and with no explanation, this closure is somewhat puzzling. I am today experiencing exactly the same issue, and came here to learn how to fix it.

Sadly it appears I'll have to leave none the wiser.

For the record, this issue does NOT appear to be fixed to me, on any recent Python:
```
(meshtastic) sholden@overhead meshtastic % python3.8
Python 3.8.18 (default, Nov  2 2023, 16:50:59)
[Clang 14.0.3 (clang-1403.0.22.14.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> open('/dev/tty.usbserial-0001')
^CTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt
>>>

(meshtastic) sholden@overhead meshtastic % python3.11
Python 3.11.5 (main, Aug 24 2023, 15:09:45) [Clang 14.0.3 (clang-1403.0.22.14.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> open('/dev/tty.usbserial-0001')
^CTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyboardInterrupt
>>>
```
Is there possibly some significant configuration item that new users such as myself are unaware of? This is on Mac OS 13.0.1 Ventura, and the web flasher had no issues communicating with the devices when installing meshtastic.

holdenweb, 2024-05-01:
Easy fix: use the callup device instead:

```
(meshtastic) sholden@overhead meshtastic % python3.11
Python 3.11.5 (main, Aug 24 2023, 15:09:45) [Clang 14.0.3 (clang-1403.0.22.14.1)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> open('/dev/cu.usbserial-0001')
<_io.TextIOWrapper name='/dev/cu.usbserial-0001' mode='r' encoding='UTF-8'>
>>>
```