meshtastic/python #351: 1.3.27 cannot set lora.region

Douile, 2022-08-01:
# Versions
- python: 3.10
- meshtastic-python: abfcbe2a900f326cb2cc784e8bb549d55061ac83
- meshtastic-device: 1.3.27

# What happened
Using serial connection setting the region (lora.region) to EU433 failed
```shell
$ meshtastic --set lora.region 2
```

```shell
$ meshtastic --set lora.region EU433
```

The CLI reports that the value is set and the device proceeds to restart but region remained unset

# Expected outcome
The region changes to the correct value

# Additional info

Not sure if this affects just this property or if it is an issue with other settings as well (further testing needed).

Douile, 2022-08-05:
Upon further investigation, the debug output seems to show valid ADMIN packets being generated, but the serial log from the device shows no indication that these are being received.

I've tried regenerating protobufs to the same version used by the version of the firmware I was using but that didn't help.

What is also strange is --reboot (another admin packet) functions correctly.

garthvh, 2022-08-05:
Are you using the pre release CLI version from pip? This is likely a firmware bug.

Douile, 2022-08-05:
No I was using locally built version of master branch, commit mentioned above. I'm pretty sure the last version I tested from pip didn't work either but need to confirm.

The latest version of the android app configures fine so not sure if its the firmware or the CLI.

Douile, 2022-08-09:
I was being an idiot, i wasn't leaving the device long enough to update the configuration before rebooting. It is strange though I had to leave the device (TTGO + heltecv2.0) a full minute after writing the configuration to get it to persist with serial. Over wifi it was a bit faster, 10 seconds did the trick, but instant reboot still dropped the changes.

porkcharsui, 2022-08-09:
I'm not so sure this issue is resolved. Setting region via CLI appears to have no effect and does not display any error `meshtastic --set lora.region US`. Attempting to read back the setting change before or after a reboot still shows the value as `unset`. Via the iOS app 1.3.33 (72), I can successfully set region. 

python: 3.9.13
meshtastic-python: https://github.com/meshtastic/Meshtastic-python/commit/95e768efd57751037ee3f5d49ff91acd000210c4
meshtastic-device: 1.3.34

garthvh, 2022-08-09:
You may need the latest python, there were protobuf changes in the last couple of firmware releases.

pip install --upgrade meshtastic --pre

meshtastic --set lora.region 1
meshtastic --get lora.region

meshtastic -- reboot
meshtastic --get lora.region 

worked for me

porkcharsui, 2022-08-09:
I reflashed to reset the device to factory defaults and tried running meshtastic-python installed from latest commit on master.

meshtastic-python: 03aab10786c293fc638f69f04769f059a333e739
firmware_version: `firmware-tbeam-1.3.34.401b5d9.bin`

This still appears to show `lora.region` value unchanged. I can however set other properties like `owner` and `channel_url`.
```
(meshtastic) bash-5.1$ meshtastic --get lora.region
Connected to radio
lora.region: 0
Completed getting preferences
(meshtastic) bash-5.1$ meshtastic --set lora.region 1
Connected to radio
Set lora.region to 1
Writing modified preferences to device
(meshtastic) bash-5.1$ meshtastic --get lora.region
Connected to radio
lora.region: 0
Completed getting preferences
```

Per your comment @garthvh I then tried the PyPy release but the same thing happened:

```
(meshtastic) bash-5.1$ pip install --upgrade meshtastic --pre
Collecting meshtastic
  Downloading meshtastic-1.3a21-py3-none-any.whl (75 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 75.9/75.9 kB 2.1 MB/s eta 0:00:00
Requirement already satisfied: pyserial>=3.4 in /Users/derek/.local/share/virtualenvs/meshtastic-hTyoJjiu/lib/python3.9/site-packages (from meshtastic) (3.5)
Requirement already satisfied: pexpect>=4.6.0 in /Users/derek/.local/share/virtualenvs/meshtastic-hTyoJjiu/lib/python3.9/site-packages (from meshtastic) (4.8.0)
Requirement already satisfied: pyqrcode>=1.2.1 in /Users/derek/.local/share/virtualenvs/meshtastic-hTyoJjiu/lib/python3.9/site-packages (from meshtastic) (1.2.1)
Requirement already satisfied: timeago>=1.0.15 in /Users/derek/.local/share/virtualenvs/meshtastic-hTyoJjiu/lib/python3.9/site-packages (from meshtastic) (1.0.15)
Requirement already satisfied: pyyaml in /Users/derek/.local/share/virtualenvs/meshtastic-hTyoJjiu/lib/python3.9/site-packages (from meshtastic) (6.0)
Requirement already satisfied: protobuf>=3.13.0 in /Users/derek/.local/share/virtualenvs/meshtastic-hTyoJjiu/lib/python3.9/site-packages (from meshtastic) (4.21.4)
Requirement already satisfied: tabulate>=0.8.9 in /Users/derek/.local/share/virtualenvs/meshtastic-hTyoJjiu/lib/python3.9/site-packages (from meshtastic) (0.8.10)
Requirement already satisfied: pypubsub>=4.0.3 in /Users/derek/.local/share/virtualenvs/meshtastic-hTyoJjiu/lib/python3.9/site-packages (from meshtastic) (4.0.3)
Requirement already satisfied: dotmap>=1.3.14 in /Users/derek/.local/share/virtualenvs/meshtastic-hTyoJjiu/lib/python3.9/site-packages (from meshtastic) (1.3.30)
Requirement already satisfied: ptyprocess>=0.5 in /Users/derek/.local/share/virtualenvs/meshtastic-hTyoJjiu/lib/python3.9/site-packages (from pexpect>=4.6.0->meshtastic) (0.7.0)
Installing collected packages: meshtastic
Successfully installed meshtastic-1.3a21
(meshtastic) bash-5.1$ meshtastic --get lora.region
Connected to radio
lora.region: 0
Completed getting preferences
(meshtastic) bash-5.1$ meshtastic --set lora.region 1
Connected to radio
Set lora.region to 1
Writing modified preferences to device
(meshtastic) bash-5.1$ meshtastic --get lora.region
Connected to radio
lora.region: 0
Completed getting preferences
```

Douile, 2022-08-09:
I think it might be firmware issue with serial connections, using wifi it worked fine. Over serial had to set region wait a minute the restart the device and it took effect.

porkcharsui, 2022-08-09:
> I think it might be firmware issue with serial connections, using wifi it worked fine. Over serial had to set region wait a minute the restart the device and it took effect.

Are you saying you had success with serial? I tried waiting 90 seconds and then rebooting but didn't seem to see it stick ever. iOS application had no issue setting this value.

```
(meshtastic) bash-5.1$ meshtastic --set lora.region 1 && sleep 90
Connected to radio
Set lora.region to 1
Writing modified preferences to device
(meshtastic) bash-5.1$ meshtastic --get lora.region
Connected to radio
lora.region: 0
Completed getting preferences

(meshtastic) bash-5.1$ meshtastic --reboot
Connected to radio
INFO file:node.py reboot line:493 Telling node to reboot in 10 seconds
(meshtastic) bash-5.1$ meshtastic --get lora.region
Connected to radio
lora.region: 0
Completed getting preferences
```

Douile, 2022-08-09:
Serial did work but questionably, I was using the master branch and I believe I also updated protobufs. The android app also works completely fine but the web app does have issues with serial as well which makes me wonder whether it is something to do with linux.

porkcharsui, 2022-08-09:
@Douile Your last comment made go back and double check that I pulled the correct proto submodule and from latest commit https://github.com/meshtastic/Meshtastic-python/commit/03aab10786c293fc638f69f04769f059a333e739. I repeated the previous tests but have the same outcome.

All my test performed on macOS 12.4, and previously meshtastic 1.2 and same macOS version previously worked. 

Is there a way to simultaneous output raw debug logs over serial, while setting this value so I can debug from the firmware side?

mkinney, 2022-08-10:
Looks like you're using bash from Windows. Can you try again using just a command prompt? (not a power shell prompt either)

porkcharsui, 2022-08-10:
> Looks like you're using bash from Windows. Can you try again using just a command prompt? (not a power shell prompt either)

I'm am using macOS 12.4. You're likely seeing my use of [pipenv](https://pipenv.pypa.io/en/latest/) to manage my virtualenv for python.

Douile, 2022-08-10:
@porkcharsui 
> Is there a way to simultaneous output raw debug logs over serial, while setting this value so I can debug from the firmware side?

Using `--seriallog seriallog.txt` will output the seriallog to a text file while CLI is running, or just using tail: `tail -f /dev/ttyUSBXX`. I would recommend observing the results for a working version of the firmware then the latest version and comparing.

Since we haven't actually confirmed whether this is a problem with the python library itself or the firmware I'm going to re-open the issue to encourage further collaboration here.

garthvh, 2022-08-10:
Both my tbeams work from both the CLI and iOS
tlora v1 does not ever save from the CLI, works from iOS, get works from CLI after setting it on iOS
tlora v2 does not ever save from the CLI, works from iOS, get works from CLI after setting it on iOS

Rak device worked from both the CLI an iOS

garthvh, 2022-08-10:
Tbeams have now lost their region after reset, so CLI set region values have held.

Douile, 2022-08-11:
When watching serial logs there are no logs related to receiving an admin packet when the cli requests to change region, but when you change via bluetooth (from android app) it reports receiving admin packets.

I have a few theories:
- CLI is sending packets with incorrect addresses so node ignores them
  - Probably not this as it works fine when using HTTP
- Serial is being configured wrong e.g. baud rate or something is miss-configured causing packets in serial to be corrupted or lost (or some other reason for usb serial as communication medium not working)
- CLI is sending invalid admin protobufs

garthvh, 2022-08-11:
This is what an acked admin message sent from iOS looks like in the serial logs

14:24:04 27417 PACKET FROM PHONE (id=0x09e2d2b9 Fr0x00 To0xf0, WantAck1, HopLim0 Ch0x0 Portnum=6 priority=70)
14:24:04 27417 Enqueued local (id=0x09e2d2b9 Fr0x00 To0xf0, WantAck1, HopLim0 Ch0x0 Portnum=6 rxtime=1660227844 priority=70)
14:24:04 27417 [Router] Add packet record (id=0x09e2d2b9 Fr0x00 To0xf0, WantAck1, HopLim0 Ch0x0 Portnum=6 rxtime=1660227844 priority=70)
14:24:04 27417 [Router] handleReceived(REMOTE) (id=0x09e2d2b9 Fr0x00 To0xf0, WantAck1, HopLim0 Ch0x0 Portnum=6 rxtime=1660227844 priority=70)
14:24:04 27417 [Router] Module 'Admin' wantsPacket=1
14:24:04 27417 [Router] Received Admin from=0x0, id=0x9e2d2b9, portnum=6, payloadlen=6

thebentern, 2022-08-11:
Please try with the latest pre-release:
`pip install --upgrade meshtastic --pre`
I made some significant changes to how we write config prefs that should make things more reliable

porkcharsui, 2022-08-11:
Latest pre-release `meshtastic-1.3a23` solved the issues I reported when flashed with `firmware-tbeam-1.3.34.401b5d9.bin`

```
(meshtastic) bash-5.1$ meshtastic --get lora.region
Connected to radio
lora.region: 0
Completed getting preferences
(meshtastic) bash-5.1$ meshtastic --set lora.region 1
Connected to radio
Set lora.region to 1
Writing modified preferences to device
(meshtastic) bash-5.1$ meshtastic --get lora.region
Connected to radio
lora.region: 1
Completed getting preferences
(meshtastic) bash-5.1$ meshtastic --reboot
Connected to radio
INFO file:node.py reboot line:532 Telling node to reboot in 10 seconds
(meshtastic) bash-5.1$ meshtastic --get lora.region
Connected to radio
lora.region: 1
Completed getting preferences
```

garthvh, 2022-08-11:
This is fixed in the latest pre-release version

pip install --upgrade meshtastic --pre

closed 2022-08-11 by garthvh