meshtastic/python #358: Warning: wifi_password must be at least 8 characters

scarolan, 2022-08-12:
The docs suggest that you should set wifi_ssid and wifi_password to empty strings to disable wifi.

But when you attempt to do that this error pops up:

```
Warning: wifi_password must be 8 or more characters.
```

This is confusing for new users.  Maybe a separate setting to toggle whether wifi is enabled might be better.

garthvh, 2022-08-12:
Those docs are for 1.2

https://meshtastic.org/docs/settings/config/wifi#disable-wifi

closed 2022-08-12 by garthvh

scarolan, 2022-08-12:
Sorry.  it wasn't clear to me that those numbers on the left were versions...

[image]

scarolan, 2022-08-12:
Also, your FAQ still points at the 1.2 docs

https://meshtastic.org/docs/faq/wifi