meshtastic/python #362: Station G1 device appears to not have ability to create a fixed location for LON/LAT/ALT

giglioroninomicon, 2022-08-24:
Giglio — Today at 9:24 AM
As an aside I went to configure the Station G1 with fixed LAT/LON and received the following when trying to do it by command line "Connected to radio
Aborting due to: Protocol message LocalConfig has no "fixed_position" field." Not sure if that is an issue on my side or something that needs to be enabled for the Station G1, just fyi. (edited)

thebentern — Today at 9:31 AM
I don't think that one ever got wired up to its new location in the config. If you don't mind, create an issue in the meshtastic-python github repo

Neil — Today at 9:33 AM
This generally means, you are on the boat of 1.3.

thebentern — Today at 9:34 AM
Right. In 1.3, we moved the position prefs into the PositionConfig
For the python cli those special commands like --setlat also set position mode to fixed I believe

Giglio — Today at 9:37 AM
FWIW, I was using the command "meshtastic --port COM6 --setalt 70" when I received the above error.

https://discord.com/channels/867578229534359593/930639909998428220/1012036562592207012

garthvh, 2022-08-24:
The --set and --get methods are working for the new config sections (config and module) but some of the previous convenience methods have not been updated in the python CLI, this issue affects all hardware.

giglioroninomicon, 2022-08-24:
Assuming I'm not missing the point, and this may be the wrong place for the question - if --set and --get work for the "new config sections (config and module)" , what would be the proper cli command to store a fixed location for a non-gps equipped device? (And a big thanks for all the great work!)

closed 2023-03-19 by garthvh