meshtastic/python #364: 1.3 Convenience Methods - Current Status

garthvh, 2022-08-26:


thebentern, 2022-08-27:
**ch-set**
Appears to work with the latest fix. My testing:
```
ch-set psk 0x1a1a1a1a2b2b2b2b1a1a1a1a2b2b2b2b1a1a1a1a2b2b2b2b1a1a1a1a2b2b2b2b --ch-index 0
ch-set psk random --ch-index 0
ch-set psk none --ch-index 0
ch-set channel_num 46 ch-index 0
```

thebentern, 2022-08-27:
Almost all of the channel methods work now. Some of them bark with this, but it doesn't appear to affect the results:
```
meshtastic --ch-add 2ndLife
ERROR file:stream_interface.py __reader line:171 Error while handling message from radio argument of type 'NoneType' is not iterable
Traceback (most recent call last):
  File "c:\python38\lib\site-packages\meshtastic\stream_interface.py", line 169, in __reader
    self._handleFromRadio(self._rxBuf[HEADER_LEN:])
  File "c:\python38\lib\site-packages\meshtastic\mesh_interface.py", line 537, in _handleFromRadio
    self._handlePacketFromRadio(fromRadio.packet)
  File "c:\python38\lib\site-packages\meshtastic\mesh_interface.py", line 678, in _handlePacketFromRadio
    if "portnum" not in decoded:
Connected to radio
```

Bergstern, 2022-09-08:
perhaps i could help if in the cli the confirming message is not only that u have set a "fixes pos", but also the message that gps is disabled?
like: 
Connected to radio
Set position.fixed_position to true
**Set position.gps_enables to false**
Writing modified preferences to device

closed 2023-03-19 by garthvh