meshtastic/python #365: Can't update time and wifi password not posting.

Commander-Crash, 2022-08-26:
Having issues setting time. Everything wants to act like it is a year behind. I have 2 LILYGO TTGO T-Beam V1.1 and one TTGO lilygo device that is managed by a raspberry pi via python. The tbeams after updating metastatic to the latest ver. 1.2.65.0adc5ce alpha from ver. 1.1.8 the tebeams no longer use on board GPS. I used GPS-SIM with a hackrf one, to try and force GPS to change but no reaction from both the tbeams. wifi password will not set with the following command:
 (meshtastic --set wifi_ap_mode false --set wifi_password "MY-PASSWORD" --set wifi_ssid WIN_502001)
It will stay as "sekrit" I don't even know what a "sekrit" is.

### meshtastic --support

If having issues with meshtastic cli or python library
or wish to make feature requests, visit:
https://github.com/meshtastic/Meshtastic-python/issues
When adding an issue, be sure to include the following info:
 System: Linux
   Platform: Linux-4.19.118+-armv6l-with-debian-10.12
   Release: 4.19.118+
   Machine: armv6l
   Encoding (stdin): UTF-8
   Encoding (stdout): UTF-8
 meshtastic: v1.2.95
 Executable: /home/pi/.local/bin/meshtastic
 Python: 3.7.3 CPython GCC 8.3.0

Please add the output from the command: meshtastic --info

###meshtastic --info
Connected to radio

Owner: Piximon-Router (Pxm)
My info: { "myNodeNum": 3186685188, "numBands": 13, "firmwareVersion": "1.2.65.0adc5ce", "rebootCount": 43, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 9.2, "airUtilTx": 1.3073611 }
Nodes in mesh:  {'num': 3186685188, 'user': {'id': '!bdf0f504', 'longName': 'Piximon-Router', 'shortName': 'Pxm', 'macaddr': '7c:9e:bd:f0:f5:04', 'hwModel': 'HELTEC_V1'}, 'position': {'latitudeI': 361033895, 'longitudeI': -958327209, 'latitude': **.****, 'longitude': -**.****}, 'lastHeard': 1609428664, 'snr': 10.75}  {'num': 84781020, 'user': {'id': '!050da7dc', 'longName': 'Xev', 'shortName': 'Xev', 'macaddr': '3c:61:05:0d:a7:**', 'hwModel': 'TBEAM'}, 'position': {'time': 1609415382}, 'lastHeard': 1609425355, 'snr': 9.75}  {'num': 2988739188, 'user': {'id': '!b2248a74', 'longName': 'Azazel', 'shortName': 'Azz', 'macaddr': 'ac:67:b2:24:8a:**', 'hwModel': 'TBEAM'}, 'position': {'time': 1609428651}, 'lastHeard': 1609428644, 'snr': 9.5}

Preferences: { "phoneTimeoutSecs": 900, "lsSecs": 300, "wifiSsid": "WIN_502001", "wifiPassword": "sekrit", "isRouter": true, "fixedPosition": true, "storeForwardPluginHistoryReturnWindow": 5, "storeForwardPluginEnabled": true, "isAlwaysPowered": true }

Channels:
  PRIMARY psk=default { "modemConfig": "Bw125Cr48Sf4096", "psk": "AQ==" }

Primary channel URL: https://www.meshtastic.org/d/#CgUYAyIBAQ

###meshtastic --node
[image]

If you need anymore info please let me know and what commands to do. Thank you.

[image]
[image]

garthvh, 2022-08-26:
1.2 is not getting updated so you may have to go down a version to get a working wifi/webui.

I don't get how you have gps configured, do you get the correct time if you use the onboard GPS modules?

Commander-Crash, 2022-08-26:
Thank you for the fast response 
I put GPS in by python command for the router device: "meshtastic --setlat **.*** --setlon -**.***" Im not to worry about the wifi or web UI I just want updated time. I tried using "meshtastic --settime" However this is not a valid command.

Commander-Crash, 2022-08-27:
I got this resolved. I'm using a lora ttgo pico now. The other lilygo ttgo device for some reason does not like the new ver of metastatic. So I will close.

closed 2022-08-27 by Commander-Crash