meshtastic/python #367: CLI Reports ModemPreset Has No Value for ch-mid command-line options

ghostop14, 2022-09-01:
meshtastic --ch-midfast and meshtastic --ch-midslow both error with:

garthvh, 2022-09-01:
/// Long Range - Fast
  case longFast // = 0
  /// Long Range - Slow
  case longSlow // = 1
 /// Very Long Range - Slow
 case vlongSlow // = 2
 /// Medium Range - Slow
 case medSlow // = 3
 /// Medium Range - Fast
 case medFast // = 4
 /// Short Range - Slow
 case shortSlow // = 5
 /// Short Range - Fast
  case shortFast // = 6
  
  Are the modem preset values from 1.3, the mid is an old typo.

ghostop14, 2022-09-02:
It may be from the command-line parsing:

ghostop14, 2022-09-02:
Looks like the mismatch is in __main__.py:

        if args.ch_midslow:
            setSimpleConfig(config_pb2.Config.LoRaConfig.ModemPreset.MidSlow)

        if args.ch_midfast:
            setSimpleConfig(config_pb2.Config.LoRaConfig.ModemPreset.MidFast)

ghostop14, 2022-09-02:
Cloned meshtastic-python and fixed and built locally. Changing those lines in __main__py to Med rather than Mid fixes the command-line issue.

garthvh, 2022-09-02:
Please submit a pull request

ghostop14, 2022-09-02:
Just submitted.

closed 2022-09-08 by thebentern