meshtastic/python #378: Installing CLI v1.3.37 breaks Warpinator on Linux Mint 21

kerastinell, 2022-10-04:
Installing Meshtastic CLI 1.3.37 prevents Warpinator (FLOSS file transfer solution made by the Linux Mint developers) from launching on Linux Mint 21 and makes it throw error messages about `protobuf`:
```
~$ warpinator
Descriptors cannot not be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).

More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates
```

It seems like there's some dependency conflict with the `protobuf` package.

Complete logs and steps to reproduce are [available in the issue](https://github.com/linuxmint/warpinator/issues/145) I opened in the warpinator repository.

garthvh, 2022-10-04:
This is the python protobufs package we are using protobuf>=3.13.0

Mint is quite a ways behind sometimes, not sure this is something we are going to be able to manage.

caveman99, 2022-10-05:
Version conflicts with python software are common, best use a venv to isolate them from each other. https://meshtastic.org/docs/software/python/python-installation (linux tab)

closed 2022-10-05 by caveman99