meshtastic/python #391: [Bug] Wifi configuration is broken

D4rk4, 2022-10-30:
**LocalConfig and LocalModuleConfig do not have attribute network.wifi_mode.**

```
meshtastic --version
1.3.44
```

Command:
`meshtastic --set network.wifi_ssid XXXX --set network.wifi_psk XXXX --set network.wifi_enabled true --set network.wifi_mode CLIENT`
Output:
```
    Connected to radio
    Set network.wifi_ssid to XXXX
    Set network.wifi_psk to XXXX
    Set network.wifi_enabled to true
    LocalConfig and LocalModuleConfig do not have attribute network.wifi_mode.
    Choices are...
    device:
        device.debug_log_enabled
        device.role
        device.serial_enabled
    position:
        position.fixed_position
        position.gps_attempt_time
        position.gps_enabled
        position.gps_update_interval
        position.position_broadcast_secs
        position.position_broadcast_smart_enabled
        position.position_flags
    power:
        power.adc_multiplier_override
        power.is_power_saving
        power.ls_secs
        power.mesh_sds_timeout_secs
        power.min_wake_secs
        power.on_battery_shutdown_after_secs
        power.sds_secs
        power.wait_bluetooth_secs
    network:
        network.eth_enabled
        network.eth_mode
        network.ipv4_config
        network.ntp_server
        network.wifi_enabled
        network.wifi_psk
        network.wifi_ssid
    display:
        display.auto_screen_carousel_secs
        display.compass_north_top
        display.flip_screen
        display.gps_format
        display.screen_on_secs
        display.units
    lora:
        lora.bandwidth
        lora.channel_num
        lora.coding_rate
        lora.frequency_offset
        lora.hop_limit
        lora.ignore_incoming
        lora.modem_preset
        lora.region
        lora.spread_factor
        lora.tx_enabled
        lora.tx_power
        lora.use_preset
    bluetooth:
        bluetooth.enabled
        bluetooth.fixed_pin
        bluetooth.mode
    mqtt:
        mqtt.address
        mqtt.enabled
        mqtt.encryption_enabled
        mqtt.json_enabled
        mqtt.password
        mqtt.username
    serial:
        serial.baud
        serial.echo
        serial.enabled
        serial.mode
        serial.rxd
        serial.timeout
        serial.txd
    external_notification:
        external_notification.active
        external_notification.alert_bell
        external_notification.alert_message
        external_notification.enabled
        external_notification.output
        external_notification.output_ms
    store_forward:
        store_forward.enabled
        store_forward.heartbeat
        store_forward.history_return_max
        store_forward.history_return_window
        store_forward.records
    range_test:
        range_test.enabled
        range_test.save
        range_test.sender
    telemetry:
        telemetry.device_update_interval
        telemetry.environment_display_fahrenheit
        telemetry.environment_measurement_enabled
        telemetry.environment_screen_enabled
        telemetry.environment_update_interval
    canned_message:
        canned_message.allow_input_source
        canned_message.enabled
        canned_message.inputbroker_event_ccw
        canned_message.inputbroker_event_cw
        canned_message.inputbroker_event_press
        canned_message.inputbroker_pin_a
        canned_message.inputbroker_pin_b
        canned_message.inputbroker_pin_press
        canned_message.rotary1_enabled
        canned_message.send_bell
        canned_message.updown1_enabled
```

caveman99, 2022-10-30:
Wifi_mode was retired recently. There's no AP any more in the software.

closed 2022-10-30 by caveman99

D4rk4, 2022-10-30:
@caveman99 need to reflect it in documentation

caveman99, 2022-10-30:
@D4rk4 absolutely right, that change was only a few days ago. The auto-generated part of the protobuf documentation doesn't have it any more, the manually edited parts were not updated yet.

garthvh, 2022-10-30:
There is an open issue in the docs already https://github.com/meshtastic/Meshtastic/issues/452

charlesmarseille, 2022-10-30:
> @caveman99 need to reflect it in documentation

you can always edit the docs and add a note saying from version 1.3.XX AP mode is dropped, thus no more mode switch.

caveman99, 2022-10-30:
> > @caveman99 need to reflect it in documentation
> 
> you can always edit the docs and add a note saying from version 1.3.XX AP mode is dropped, thus no more mode switch.

Funny thing, i did exactly this a few hours back. But you are right, not many people notice the edit link on these pages

charlesmarseille, 2022-10-30:
Great! The more testers and writers, the more happy users.