meshtastic/python #393: Proposition: add description and details of set variables in CLI

charlesmarseille, 2022-10-31:
Instead of refering to the online docs each time we work in the CLI, typing meshtastic --get 0 could output the following structure:

display:
    display.auto_screen_carousel_secs        # time between screen pages rotation (seconds)
    display.compass_north_top                   # impose compass orientation (bool)
    display.flip_screen                                 # orientation of oled screen (bool)
    display.gps_format                                # format of lat, lon coordinates ("gpsFormatDec", "gpsFormatDms", ...)  
    display.screen_on_secs                          # Time to keep screen on (seconds)
    display.units                                           # .....

Having the description, setting value type and possible values and exceptions would be useful.

garthvh, 2022-10-31:
Great idea, feel free to send in a pull request

charlesmarseille, 2022-10-31:
I'll give a first draft, but some settings I have no idea what values they take and are not described in the docs. I'll have to search the code. I'll give it a go and ask for completion from you guys.